Ultimaker/Cura #92: Loading objects is *much* slower than in legacy-Cura

fieldOfView, 2015-07-03:
Loading the Ultimaker Robot takes around 12 seconds with 15.06.01, versus 2 seconds with 15.04. A factor of 6 slower. I have not even tried loading a more complex object (eg a couple million polygon 3d scan...).

Note that slicing also seems slower, but not by as much; After loading the Ultimaker Robot, it takes the new version another 12(ish) seconds to slice the object, versus 4 seconds on legacy-Cura. This may be due to new features in the slicer, and the feeling of slowness is compounded by what looks like a pause before the background slicing process kicks in.

BagelOrb, 2015-07-03:
The slow model loading certainly is a bug. It shouldn't take longer in the new interface.

New features which might slow down the new CuraEngine:
- remove overlapping wall : pieces of walls which would result in overlapping parts in the lines laid down are removed
- fill wall gaps : the gaps created by the removed walls (where there is overlap) are filled
- coasting : the last piece of a path before a travel move through air is printed using the material which would otherwise be oozed
- avoid other parts : during a travel through air the print head avoids other parts, so that it has less chance to collide with them
- combing is enhanced, combing is now also performed when moving to outside an object
- the support system has been totally revised. The areas where support is placed are more accurate and more such areas are generated.

fieldOfView, 2015-07-03:
I forgot to mention my system specs:
Windows 7 64 bit, Core i5 @ 2.3Ghz, 8 Gb.

nallath, 2015-07-03:
We have an idea what causes the model slowing to occur; it might have something to do with the (revised) normal generation.

closed 2015-07-08 by awhiemstra