Ultimaker/Cura #93: Layer slider has poor visibility

fieldOfView, 2015-07-03:
The slider used to select the layer to view uses the same color as the printer bed checkerboard and has poor visibility when the top layer is shown. The Layer view opens with the top layer visible by default. Depending on the view of the printerbed, it is easily possible to obscure the slider almost entirely.

Switching the backgroundcolor of the slider to the color the buttons have would increase visibility (though those buttons could do with a bit more contrast from the checkerboard grey as well).

nallath, 2015-07-03:
Thanks for reporting. If you want a quick fix, you can edit the theme.json file (in the resoureces folder). If I remember correctly you need to change the value of "slider_groove". The color is defined by RGBA.

fieldOfView, 2015-07-03:
Actually, it is slider_groove_border and slider_groove_fill; the light part of the slider is set with slider_groove. Would a pull-request be appreciated?

nallath, 2015-07-03:
Yup. Thats why we open source things :)

fieldOfView, 2015-07-03:
@nallath: against which branch? Master or 15.06?

nallath, 2015-07-06:
FYI: Master is the branch where we continue to commit / create new features. 15.06 is the release branch for bugfixes only.

closed 2015-07-06 by nallath