Ultimaker/Cura #96: printer selection crashes on Windows 7

TinkerGnome, 2015-07-03:
It's a german Windows 7 and the very first dialog leads to an application crash.
"Run as administrator" shows the same behavior.

[image]

```
Problemsignatur:
Problemereignisname: APPCRASH
Anwendungsname: Cura.exe
Anwendungsversion: 0.0.0.0
Anwendungszeitstempel: 536d0024
Fehlermodulname: StackHash_0a9e
Fehlermodulversion: 0.0.0.0
Fehlermodulzeitstempel: 00000000
Ausnahmecode: c0000005
Ausnahmeoffset: 00000000
Betriebsystemversion: 6.1.7601.2.1.0.256.48
Gebietsschema-ID: 1031
Zusatzinformation 1: 0a9e
Zusatzinformation 2: 0a9e372d3b4ad19135b953a78882e789
Zusatzinformation 3: 0a9e
Zusatzinformation 4: 0a9e372d3b4ad19135b953a78882e789 
```

cura.log is inconspicuous

```
2015-07-03 14:23:25,135 - INFO - Loaded plugin FileLogger
2015-07-03 14:23:25,860 - INFO - Loaded plugin LocalFileStorage
2015-07-03 14:23:25,887 - INFO - Loaded plugin LayerView
2015-07-03 14:23:25,929 - INFO - Loaded plugin MeshView
2015-07-03 14:23:25,973 - INFO - Loaded plugin WireframeView
2015-07-03 14:23:25,990 - INFO - Loaded plugin OBJReader
2015-07-03 14:23:25,999 - INFO - Loaded plugin STLReader
2015-07-03 14:23:26,020 - INFO - Loaded plugin STLWriter
2015-07-03 14:23:26,035 - INFO - Loaded plugin GCodeWriter
2015-07-03 14:23:26,066 - INFO - Loaded plugin CameraTool
2015-07-03 14:23:26,094 - INFO - Loaded plugin MirrorTool
2015-07-03 14:23:26,157 - INFO - Loaded plugin RotateTool
2015-07-03 14:23:26,189 - INFO - Loaded plugin ScaleTool
2015-07-03 14:23:26,219 - INFO - Loaded plugin SelectionTool
2015-07-03 14:23:26,252 - INFO - Loaded plugin TranslateTool
2015-07-03 14:23:26,296 - INFO - Checking for new version of cura
2015-07-03 14:23:26,296 - INFO - Loaded plugin UpdateChecker
2015-07-03 14:23:26,347 - INFO - Loaded plugin USBPrinting
2015-07-03 14:23:26,351 - INFO - Could not establish connection on COM13: 'Failed to open serial port'. Device is not arduino based.
2015-07-03 14:23:26,355 - INFO - Loaded plugin CuraEngineBackend
2015-07-03 14:23:27,952 - DEBUG - Loaded theme C:/Program Files/Cura_15.06.01/resources/themes/cura
2015-07-03 14:23:28,461 - INFO - Started engine process: C:\Program Files\Cura_15.06.01\CuraEngine.exe
2015-07-03 14:23:31,730 - DEBUG - Backend connected on port 49674 
```

feiteira, 2015-07-06:
Same issue here, it appears to work with other printers, except the "Ultimaker 2".

nallath, 2015-07-06:
So only selecting UM2 causes the crash?

feiteira, 2015-07-06:
I didn't try all options, so I cannot say exactly, at leas the Ultimaker Original was not crashing.

I've downgraded to 15.06.00 and the problem was still there.
I've re-downgraded to 15.04 and the problem was gone.

nallath, 2015-07-06:
15.04 is completely different from 15.06, so it's logical that the problem is gone.

awhiemstra, 2015-07-06:
Can you please try the Ultimaker Original+ ? I have a suspicion that it is a driver bug related to texture uploads that crashes, in which case the UMO+ should also crash.

TinkerGnome, 2015-07-06:
The same behavior for the Original+, the only working machine selection is "Ultimaker Original"
but you're on the right track...

awhiemstra, 2015-07-10:
I have added a command line switch --disable-textures that disables texture loads. It will be available for the next patch release. Once that is available, please test to see if Cura does not crash. If that works, we can look at why texture loads crash, but I suspect it is an upstream bug.

HetHooghuis, 2015-07-10:
Same problem over here. It only works when selecting the "Ultimaker Original"

TinkerGnome, 2015-07-10:
tested with: "...\Cura_15.06.02\Cura.exe" --disable-textures"

result:  "Ultimaker2" is selectable without crash (as predicted...)

I wonder why the texture for the Ultimaker Original works flawlessly 
(even without the command line switch).
What is the difference between the textures? Could this be a starting 
point for a (temporary) workaround?

TinkerGnome, 2015-07-12:
a small addition:
I've just noticed that the "ultimaker_original.json" doesn't contain the "platform_texture" at all....
Removing this line from the "ultimaker2.json" is also a temporarily solution for me..  ;-)

awhiemstra, 2015-07-13:
The Ultimaker Original does not use any actual textures, everything is in the model or in shaders, which is why I suspected the texture uploading. Glad to know the workaround at least works, now to figure out what is causing Qt to crash on texture loads...

HetHooghuis, 2015-09-08:
The new (15.06.03) softwarerelease brings no solution for me!
unfortunately adding the  --disable-textures paramater also doesn't work for me? Reading the other comments I think this is very strange?  
When is a solution to be expected?

nallath, 2015-09-08:
Could you check the 15.10 beta? You can find it at software.ultimaker.com/Cura_closed_beta/15.10

HetHooghuis, 2015-09-10:
Thanks for your reply..  Unfortunately this is also not working. I think it has to do with an outdated computer. It's an old Lenovo.. When replacing it by a newer HP computer it works!

nallath, 2015-09-10:
Yeah, that also makes sense. Graphics cards can be difficult beasts to work with.

TinkerGnome, 2015-09-12:
I can confirm that the problem is **not solved**. 15.10 beta is still unusable with enabled textures.
(Acer TravelMate with integrated Intel HD Graphics)

nallath, 2015-09-12:
Allright, we will have a look at it again. Thanks for testing it!

Ghostkeeper, 2015-09-13:
The bug does not occur on my Windows 7 laptop, Lenovo Y580, which has an NVIDIA GeForce GTX660M graphics card. I can just select the Ultimaker 2 printer and everything works as expected. This is version 15.09.81.

awhiemstra, 2015-09-13:
We found some hardware at the office that also exhibits this issue. This means we might be able to find out what the actual issue is and possibly file an upstream bug, since I am pretty sure this is an issue in either Qt or even the graphics driver.

diracsbracket, 2016-03-02:
Hi
I found posts about the issue before and tried the --disable-textures. But that doesn't work in the development version of Cura. Since I don't know how to pass that launch argument, I followed the same procedure as described in another post, i.e. modified the Windows shortcut to the Cura executable and added the  --disable-textures switch there.

Also note thate I successfully built Cura using the Daid script, and that I don't have that crash with that version. All the posts relating to this crash seem to be from people not using the development version of Cura, which I am referring to here.

fieldOfView, 2016-03-02:
@diracsbracket, could you be more specific about the version(s) you are talking about? Which "development version" have you tested? 15.06.03, 15.10, or an unreleased 2.1 build?

closed 2016-09-17 by thopiekar

diracsbracket, 2016-10-25:
It's a bit late for this reply, but it was for Cura 2.0.99....