Ultimaker/Cura #105: Custom file writer is not use even though it is selected in 15.06.01

gearsawe, 2015-07-04:
Have a few problems here. When I have the a custom gcode writer installed in the plugins folder I cannot uncheck it. second even though it is check it does not seem to be used. or It may be doing be something I'm not doing correct. (first issue posted on github)

Place the following to scripts in the plugins folder and name it "GCodeWriterM3D". In this example the output should only have the text "Custom_GCODE"

Script: 
```
__init__.py
```
```
from . import GCodeWriterM3D

def getMetaData():
    return {
        "type": "mesh_writer",
        "plugin": {
            "name": "GCode Writer for M3D",
            "author": "Joe Brown",
            "version": "1.0",
            "description": "Writes GCode to a file M3D format"
        },

        "mesh_writer": {
            "extension": "gcode",
            "description":"GCode File M3D"
        }
    }

def register(app):
    return { "mesh_writer": GCodeWriterM3D.GCodeWriterM3D() }
```

and the other script:
```
GCodeWriterM3D.py
```
```
from UM.Mesh.MeshWriter import MeshWriter
from UM.Logger import Logger
from UM.Application import Application
import io

class GCodeWriterM3D(MeshWriter):
    def __init__(self):
        super().__init__()
        self._gcode = None

    def write(self, file_name, storage_device, mesh_data):
        if "gcode" in file_name:
            scene = Application.getInstance().getController().getScene()
            gcode_list = getattr(scene, "gcode_list")
            if gcode_list:
                f = storage_device.openFile(file_name, "wt")
                Logger.log("d", "Writing GCode to file %s", file_name)
                f.write("Custom_GCODE")
                storage_device.closeFile(f)
                return True

        return False
```

awhiemstra, 2015-07-06:
The problem here is that the extension "gcode" is used to determine what plugin to use to write the file. Eventually I want to use mime types for this but that will take a while to implement. As a workaround for now, to get it to work, you can use a different file extension.

nallath, 2015-08-24:
If you know explicitly how the g-code needs to be different, you could also use the postProcessing plugin or see if you can get this to work with an output device plugin.

closed 2015-08-24 by nallath