Ultimaker/Cura #106: Non-circular bed represented as square

quillford, 2015-07-05:
If ```"machine_center_is_zero"``` is true, it is safe to assume the bed shape is circular.

Ghostkeeper, 2015-07-05:
I think it'd be good to implement a feature such as round beds (Cura's support for non-Ultimaker machines isn't great, since it's not much of a priority, as you can imagine) but when it is implemented, the shape of the bed should be a separate parameter. I imagine there likely are printers out there with square beds but zero in the centre. Or else there will be in the future.

gearsawe, 2015-07-05:
And soon there will be a need for print beds with triangle shaped beds with rounded corners. See Tiko on kickstater.
Their plan is to use every square in of the print bed of a delta printer.

This can be sort of done now but will take lots of manual entries in the "machine_disallowed_areas" shapes can only convex not concave. So it would take many quads or triangle to fill in the areas where it cannot print.

BagelOrb, 2015-07-06:
It only takes one person many manual entries. The json file can then be included in the distribution via a pull request.

nallath, 2015-07-06:
The machine_disallowed_areas is not used when scaling to max. We could use it as a quick fix, but it should be done properly some time in the future.

nallath, 2015-07-08:
Closing this issue, as I dont believe that machine_center_is_zero should have anything to do with bed size.

closed 2015-07-08 by nallath