Ultimaker/Cura #107: Slow layer view

luisonoff, 2015-07-05:
When in layer view, with a very simple model, moving the slider to change layers adds lots of lag and  its response is very slow. It was much more fluid in 15.04, same PC same model.
Also no layer number is being shown and there is no option to change between showing all layers or just one, which I loved.

nallath, 2015-07-06:
It should be fast, but the fact that it worked in 15.04 is no guarantee that it works for this version as it's completely different in how it does everything.

irrenhaus, 2015-07-06:
+1 from my side, the layer view very slow (on Windows 7, Core i5, GTX560) while it was way faster in 15.04.
Anything I can do to help debugging?

nallath, 2015-07-06:
Not at the moment. If we have a solution / patch / hotfix we will notify you guys.

awhiemstra, 2015-07-06:
Actually, one thing that would help is providing information about what graphics chip you are using. Additionally, try updating your graphics drivers.

luisonoff, 2015-07-06:
I have Windows 8.1 64bits, GPU HD7700, tested with drivers 15.5 and 15.6. Same slow performance.

I only notice this slowness when model has large horizontal area, even with very simple models.

For example I load a big cilinder, wait for it to slice, go to layer view, move layer slider from one end to the other, it takes almost 2 seconds to show new layer and slider also goes slow, I mean I move the mouse and 2 seconds later the slider moves and layer appears.

luisonoff, 2015-07-22:
Just wanted to remind that this is still hapenning in version 15.06.2. Maybe it has improved a bit, but it is still very laggy.

nallath, 2015-09-23:
Is it stil laggy for 15.09.xx?

luisonoff, 2015-09-23:
Where can I download Cura version 15.09?

nallath, 2015-09-23:
http://software.ultimaker.com/Cura_closed_beta/15.10/

closed 2016-09-19 by thopiekar