Ultimaker/Cura #108: [Improvement] Change UI text size

luisonoff, 2015-07-05:
In my screen (1920*1080, 23 inches), some of the text displayed is too small, like text displayed inside progress bar.

I would love an option in settings, for example, to scale UI text size, to make it bigger and more easy to read in my case.

awhiemstra, 2015-07-07:
We currently scale the UI text based on system font size. However, if certain bits of text are too small, we probably need to set a minimum font size. Can you provide screenshots of the things you consider too small?

luisonoff, 2015-07-07:
Menu text and settings text is small but OK, but look at the progress bar, the save button and top text where you have to choose mode and machine.

[image]

TMHogenhout, 2015-08-07:
Duplicate. See: https://github.com/Ultimaker/Cura/issues/194#issuecomment-128656307

closed 2015-08-07 by TMHogenhout

bear454, 2017-04-13:
The font size comes from the Qt libraries... you can override by setting an environment variable: `QT_SCALE_FACTOR`

For example, on Linux, I run Cura with the command `QT_SCALE_FACTOR=2 Cura-2.4.0.AppImage`

tfischer4765, 2017-11-05:
QT_SCALE_FACTOR=2 may be a wee bit much, I'd recommend 1.1 or 1.2