Ultimaker/Cura #109: [Improvement] Disable automatic slicing

luisonoff, 2015-07-05:
I find quite disturbing that every time I change some setting, Cura slices the model. This slicing slows down everything and with complex models takes a long time. 

Sometimes I have to make many changes before slicing once I load a new model. I would love an option in settings to disable this automatic slicing on every change, and let me decide when I want to slice, for example changing progress bar to a button I can click once I am finished changing settings.

gearsawe, 2015-07-05:
So would you suggest once auto slicing is off that a button would appear next the rotate, mirror, and scale buttons.

Dim3nsioneer, 2015-07-06:
Auto-Slicing stops if you move the object out of the valid range, i.e. when it becomes gray...then you have all the time you need for changing settings.

luisonoff, 2015-07-06:
@Dim3nsioneer - Thanks for the tip, but still this does not solve cases where I have multiples STLs being printed and I don't want to lose bed arrangement.

@gearsawe - I think best option would be that once you make a setting change that needs reslicing, the progress bar where printing time is shown  changes to a clickable button to Slice again the model, so we know when it is already sliced (time result shown) or when we have to slice. If this option is not easy to implement I would put the slice button next to the "open stl" button, but then it should change color (for example) to know when it is sliced or need to be sliced.  Also, the save button should not be clickable until it is sliced with newest settings.

quillford, 2015-07-06:
At least for the old Cura, auto slicing would freeze Cura after every change if using a post-processing plugin like [OctoUpload](https://github.com/quillford/OctoUpload) because Cura would wait for it to upload each time it sliced.

Fass-ohne-Hoden, 2015-07-07:
I want to second the request. The program should
# not trigger a refresh automatically, but just indicate that things have changed and refreshing / new slicing may, therefore, be required and
# in the meantime, stop the user from doing stupid things (as saving files and stuff).

If the developers want to keep the auto slicing feature as default behaviour for new users, at least it should be disable'able (switch-offable?)

Fass-ohne-Hoden, 2015-07-07:
I want to second the request. The program should
- not trigger a refresh automatically, but just indicate that things have changed and refreshing / new slicing may, therefore, be required and
- in the meantime, stop the user from doing stupid things (as saving files and stuff).

If the developers want to keep the auto slicing feature as default behaviour for new users, at least it should be disable'able (switch-offable?)

gearsawe, 2015-07-07:
I third the request.
I have added my own custom settings to a custom printer, and every time I change something there it updates the slicing even though a setting does not actually change anything.

Dim3nsioneer, 2015-07-07:
Guys... just for your information: this discussion was going on before... this was the result:
http://oldforum.ultimaker.com/index.php?/topic/8574-cura-autoslicing-poll/?hl=%2Bautoslicing+%2Bpoll

nallath, 2015-07-07:
Well, they are right about changing a setting that does not affect the slicing (such as temperature). In that case the slicing should not be re-done.

Fass-ohne-Hoden, 2015-07-07:
As for the old thread on that topic ...
Maybe I got it wrong at that time (I didn't read all of it and of what I did read I remember only bits), but I think it was like that:
- Majority of people wanted at least an option to disable auto slicing
- Daid said something like: no need for that, all the hiccups and the sloooow process of whatever-happens-while-the-progress-bar-turns-from-empty-to-full have nothing to do with the slicing at all ... and that with the new Cura everything will be so much faster, smoother, better and so on ...
Please correct me if I'm wrong (and I expect to be to some extend).

However, here it is, the new Cura and whatever-happens-while-the-progress-bar-turns-from-empty-to-full still takes loads of time. It's become even slower than with the old Cura. Perhaps we could call the process "Refresh" instead of "Slicing" because people do not care about what it exactly is that happens in the background again and again, they're just annoyed by it.

So, unless Cura doesn't manage to get this whatever-it-is done within an unnoticeably short(!) amount of time, could we please have a setting to disable auto refresh?

gearsawe, 2015-07-07:
So if there is a setting that is specific to a printer and does not affect the slicing why does it have to reslice. Or I have no idea how to make some custom printer and gcode writer So when I save the gcode the gcode will be correct for that printer.

Dim3nsioneer, 2015-07-08:
@nallath If you really make Cura not reslicing after a temperature change you might want to add some other cases where no reslicing is needed, e.g. a simple translation in the x-y-plane (just recalculating the coordinates).

nallath, 2015-07-08:
Well first off; The -vocal- majority of people said that they wanted it. Thats a big difference from the majority of people. I did not hear thousands of users complain about it, only at that point would it be the majority.

Why we at that point in time refused to remove it is because all arguments about why it should be removed were shamanistic (eg; The problems 'caused' by automatic slicing were not caused by slicing at all, but people thought this as this is how the human mind works). It wasn't just Daid saying it, it was all the software engineers that said it.

I'm not adverse to removing the auto slicing, but have any of you noticed that it actually slices the object three times when you press slice? It does a ton of background work without even showing this to you. I think the slower slicing has something to do with layer view, so the slowdown will be gone soon. 

The slicer now always re-slices when a setting has been changed and this is bad behavior.

awhiemstra, 2015-07-08:
The slowdown was due to always processing the layer view data in the background. I have now added a check to make sure it only does this when the layer view is active.

As for not reslicing after a temperature change, that would mean the engine would need to send the GCode with template markers for certain settings that we would replace in a post-processing step, otherwise the GCode will not have the right temperature settings. Considering the amount of work this would be versus the minor gain I am not really willing to implement that.

@Dim3nsioneer while it may seem like a simple translation, all the GCode would have to be rewritten anyway since that is in absolute coordinates, meaning it would be something we could try and improve on the engine side but the frontend can do little about it.

In general I think it is better to focus on improving the slicing speed and other things relating to auto-slicing rather than adding something that is only a workaround.

awhiemstra, 2015-07-08:
Additionally, what I would not strictly oppose, is adding an option to only slice when on the layer view or when saving GCode. This would introduce a delay while saving and thus would not be the default but should cover most of the use cases mentioned here.

luisonoff, 2015-07-08:
I do think that autoslicing should NOT be removed and it should be de default behaviour. 

I am just proposing to add an option in settings so some users can disable it and slice just when they feel like. awhiemstra idea of only slicing when you click save is good, but that way we have no idea about printing time nor how much material is used, and I find those values very useful before deciding it is ready to save.

Fass-ohne-Hoden, 2015-07-08:
When working ("working" :D) with Cura, the layer view is pretty much the only view I use (except for duplicating/moving/rotating stuff). So, "it'd only happen in layer view" is no argument. There are two ways out of this, I think
- give an option to disable auto-whatever and add a nice "Refresh" button, e.g. as described earlier here
- make the slowdown go away altogether (even with active layer view)

There is no point in arguing whether actually the slicing is the problem or the doing all the graphics magic because this difference doesn't matter (from a user perspective).

That all said, I think it's a good thing to try to find the root of a problem instead of just tinkering on the symptoms. HOWEVER, this is an idealistic approach that just doesn't work out in every detail of a complex real life system, NEVER EVER. It just doesn't. Among other things, this separates pragmatist from theorists - to know where and when a workaround or a dirty fix is appropriate (be it only temporary) ...

nallath, 2015-07-09:
We've already done some improvements that should tackle the slowdown, so i'd like to find out if that is enough before hacking / hotgluing fixes ;)

qmr, 2015-07-25:
This needs to be an option.  Why is there debate about this?

I am working with a *very* complex model at the moment.  Slicing takes 20 minutes.

gearsawe, 2015-07-25:
Question: If slicing starts then you make a change a second later does the previous slice have to finish before the new parameters are accepted and the new slicing starts? Or does it abort the current slicing and start a new one with the new parameters?

luisonoff, 2015-07-26:
I like that automatic slicing abort when parameter is changed, problem I see is that you won't be able to touch any option while it slices.

I found a workaround for this automatic slicing, just change layer heigth to 0, it wont slice. Do every change you want, and at the end reset your layer heigth to what it was.

nallath, 2015-07-27:
It aborts the slicing. It does not have to wait until it is done. Starting over costs you nothing. You can change options while it slices.

qmr, 2015-07-27:
You can only easily change settings with simple models.  WIth complex models Cura will be effectively frozen and it takes 1-3 minutes to enter a new parameter.  

Then the CPU is spinning again slicing, and another 1-3 minutes to change another parameter.

nallath, 2015-07-28:
I've tried this. Even with models of 80 MB I notice no such thing. This means its a bug that we need to fix. Removing the auto slicing simply hides the symptoms.

closed 2015-08-14 by nallath

qmr, 2015-08-14:
Why is this being closed?  

Making my computer unusable for 20-30 minutes is not OK.

sbias, 2015-08-14:
i'd like to give this feature a +1. if preping on my notebook the autoslicing eats up my battery, iven if i know i still have to do this and that after mooving the first thing around. auto-slicing enabled by default is ok, but an option in expert to disable it would not hurt and would solve this problem ;)

nallath, 2015-08-17:
Like i said; Its not the auto slicing that makes this slow, it's some other bug. Your GUI should not slow down when slicing, as the slicing process is started at the lowest priority possible. This means that your operating system will give -any- process (other than the slicing) more resources if they ask for it. 
This means it's impossible for the GUI to slow down once the slicing starts. If it does do that, its a bug, which you can report in it's own issue. 
I do find merrit in the argument of sbias, as it does make your laptop eat more of it's batteries. But if disable automatic slicing is disabled, we might not find issues that some people are experiencing with the slowdown.

gearsawe, 2015-08-17:
agreed need a bug opened to report the problem of interface slowing way down during slicing. There have been a few comments by other to change things like set layer height to zero to disable slicing. This is at least a work around.

ChrisHeerschap, 2016-11-29:
"Your GUI should not slow down when slicing" - "should not" and actually does are two separate things. I've got a year old macbook pro and with five fairly simple models loaded, it beachballs between each operation. The rest of the computer remains usable, but it takes 10x longer to do simple movements/scaling/rearranging than it should.

Setting the layer height to 0 accomplishes the desired disabling of auto slicing and makes it usable again. Many thanks to @gearsawe for that protip.

With all the hidden advanced controls available for printing, it is kinda surprising that adding a simple option to disable automatic slicing is such a problem.