Ultimaker/Cura #110: No picture in Macbook mid 2015 - OS X 10.10.4

computerflummi, 2015-07-05:
Dears,

after importing STL file it is not visible. Even the build plate can not be seen. Previous version worked all finde on same machine.

Hope you can fix that ?

Holger

nallath, 2015-07-06:
Could you post the log file?

awhiemstra, 2015-07-06:
Please also provide information about what kind of graphics chip you are using and what OS you are running on.

computerflummi, 2015-07-06:
There is no log file cause there is no crash of the software. It simply shows nothing else as the user interface. 

System is (with latest 'OSX):

Modellname:	MacBook
Modell-Identifizierung:	MacBook8,1
Prozessortyp:	Intel Core M
Prozessorgeschwindigkeit:	1,1 
Anzahl der Prozessoren:	1
Gesamtanzahl der Kerne:	2
L2-Cache (pro Kern):	256 KB
L3-Cache:	4 MB
Speicher:	8 GB
Boot-ROM-Version:	MB81.0164.B06

computerflummi, 2015-07-06:
[image]
[image]

Please see screenshots. Part loaded = no STL file can be seen also no print platform ...

nallath, 2015-07-06:
There should be a log file, even when it didn't crash.

Xisle, 2015-07-06:
Log file is at ~/.cura/cura.log

computerflummi, 2015-07-06:
nope, I can not find it. can you give me the complete location as an example please ?

Xisle, 2015-07-06:
~ is your User folder.

I use Terminal and this command:
  nano ~/.cura/cura.log

*nano can be replaced with your text editor of choice.

computerflummi, 2015-07-06:
please rename to .log
[image]

computerflummi, 2015-07-06:
full content:

2015-07-02 18:43:04,271 - INFO - Loaded plugin FileLogger
2015-07-02 18:43:04,789 - INFO - Loaded plugin LocalFileStorage
2015-07-02 18:43:04,803 - INFO - Loaded plugin MeshView
2015-07-02 18:43:04,811 - INFO - Loaded plugin WireframeView
2015-07-02 18:43:04,824 - INFO - Loaded plugin LayerView
2015-07-02 18:43:04,829 - INFO - Loaded plugin OBJReader
2015-07-02 18:43:04,833 - INFO - Loaded plugin STLReader
2015-07-02 18:43:04,839 - INFO - Loaded plugin OBJWriter
2015-07-02 18:43:04,843 - INFO - Loaded plugin STLWriter
2015-07-02 18:43:04,850 - INFO - Loaded plugin GCodeWriter
2015-07-02 18:43:04,857 - INFO - Loaded plugin CameraTool
2015-07-02 18:43:04,869 - INFO - Loaded plugin MirrorTool
2015-07-02 18:43:04,899 - INFO - Loaded plugin RotateTool
2015-07-02 18:43:04,908 - INFO - Loaded plugin ScaleTool
2015-07-02 18:43:04,912 - INFO - Loaded plugin SelectionTool
2015-07-02 18:43:04,917 - INFO - Loaded plugin TranslateTool
2015-07-02 18:43:04,925 - INFO - Checking for new version of cura
2015-07-02 18:43:04,925 - INFO - Loaded plugin UpdateChecker
2015-07-02 18:43:04,949 - INFO - Loaded plugin USBPrinting
2015-07-02 18:43:04,973 - INFO - Loaded plugin CuraEngineBackend
2015-07-02 18:43:04,977 - INFO - Could not establish connection on /dev/cu.Bluetooth-Modem: 'Failed to open serial port'. Device is not arduino based.
2015-07-02 18:43:06,048 - ERROR - Error when loading file .DS_Store: 'utf-8' codec can't decode byte 0xff in position 578: invalid start byte
2015-07-02 18:43:06,394 - DEBUG - Loaded theme /Applications/Cura.app/Contents/Resources/cura/resources/themes/cura
2015-07-02 18:43:06,606 - INFO - Could not establish connection on /dev/cu.Bluetooth-Incoming-Port: 'Timeout'. Device is not arduino based.
2015-07-02 18:43:07,772 - INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
2015-07-02 18:43:07,782 - DEBUG - Backend connected on port 49674
2015-07-05 12:48:51,008 - INFO - Loaded plugin FileLogger
2015-07-05 12:48:51,411 - INFO - Loaded plugin LocalFileStorage
2015-07-05 12:48:51,436 - INFO - Loaded plugin MeshView
2015-07-05 12:48:51,443 - INFO - Loaded plugin WireframeView
2015-07-05 12:48:51,456 - INFO - Loaded plugin LayerView
2015-07-05 12:48:51,461 - INFO - Loaded plugin OBJReader
2015-07-05 12:48:51,465 - INFO - Loaded plugin STLReader
2015-07-05 12:48:51,469 - INFO - Loaded plugin OBJWriter
2015-07-05 12:48:51,473 - INFO - Loaded plugin STLWriter
2015-07-05 12:48:51,480 - INFO - Loaded plugin GCodeWriter
2015-07-05 12:48:51,485 - INFO - Loaded plugin CameraTool
2015-07-05 12:48:51,494 - INFO - Loaded plugin MirrorTool
2015-07-05 12:48:51,517 - INFO - Loaded plugin RotateTool
2015-07-05 12:48:51,524 - INFO - Loaded plugin ScaleTool
2015-07-05 12:48:51,530 - INFO - Loaded plugin SelectionTool
2015-07-05 12:48:51,536 - INFO - Loaded plugin TranslateTool
2015-07-05 12:48:51,543 - INFO - Checking for new version of cura
2015-07-05 12:48:51,544 - INFO - Loaded plugin UpdateChecker
2015-07-05 12:48:51,564 - INFO - Loaded plugin USBPrinting
2015-07-05 12:48:51,588 - INFO - Loaded plugin CuraEngineBackend
2015-07-05 12:48:51,596 - INFO - Could not establish connection on /dev/cu.Bluetooth-Modem: 'Failed to open serial port'. Device is not arduino based.
2015-07-05 12:48:52,839 - ERROR - Error when loading file .DS_Store: 'utf-8' codec can't decode byte 0xff in position 578: invalid start byte
2015-07-05 12:48:53,150 - DEBUG - Loaded theme /Applications/Cura.app/Contents/Resources/cura/resources/themes/cura
2015-07-05 12:48:53,232 - INFO - Could not establish connection on /dev/cu.Bluetooth-Incoming-Port: 'Timeout'. Device is not arduino based.
2015-07-05 12:48:54,972 - INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
2015-07-05 12:48:55,023 - DEBUG - Backend connected on port 49674
2015-07-05 12:49:13,449 - DEBUG - Loaded a mesh with 48 vertices
2015-07-05 12:49:34,016 - DEBUG - Loaded a mesh with 36 vertices
2015-07-05 12:49:59,627 - INFO - Checking for new version of cura
2015-07-05 21:18:40,850 - INFO - Loaded plugin FileLogger
2015-07-05 21:18:41,260 - INFO - Loaded plugin LocalFileStorage
2015-07-05 21:18:41,268 - INFO - Loaded plugin MeshView
2015-07-05 21:18:41,274 - INFO - Loaded plugin WireframeView
2015-07-05 21:18:41,283 - INFO - Loaded plugin LayerView
2015-07-05 21:18:41,286 - INFO - Loaded plugin OBJReader
2015-07-05 21:18:41,289 - INFO - Loaded plugin STLReader
2015-07-05 21:18:41,293 - INFO - Loaded plugin OBJWriter
2015-07-05 21:18:41,296 - INFO - Loaded plugin STLWriter
2015-07-05 21:18:41,301 - INFO - Loaded plugin GCodeWriter
2015-07-05 21:18:41,306 - INFO - Loaded plugin CameraTool
2015-07-05 21:18:41,316 - INFO - Loaded plugin MirrorTool
2015-07-05 21:18:41,338 - INFO - Loaded plugin RotateTool
[... 72 lines cut ...]
2015-07-06 20:41:06,673 - INFO - Loaded plugin LocalFileStorage
2015-07-06 20:41:06,681 - INFO - Loaded plugin MeshView
2015-07-06 20:41:06,685 - INFO - Loaded plugin WireframeView
2015-07-06 20:41:06,693 - INFO - Loaded plugin LayerView
2015-07-06 20:41:06,696 - INFO - Loaded plugin OBJReader
2015-07-06 20:41:06,699 - INFO - Loaded plugin STLReader
2015-07-06 20:41:06,701 - INFO - Loaded plugin OBJWriter
2015-07-06 20:41:06,704 - INFO - Loaded plugin STLWriter
2015-07-06 20:41:06,710 - INFO - Loaded plugin GCodeWriter
2015-07-06 20:41:06,714 - INFO - Loaded plugin CameraTool
2015-07-06 20:41:06,720 - INFO - Loaded plugin MirrorTool
[... cut ...]
2015-07-06 20:41:07,875 - ERROR - Error when loading file .DS_Store: 'utf-8' codec can't decode byte 0xff in position 578: invalid start byte
2015-07-06 20:41:08,120 - DEBUG - Loaded theme /Applications/Cura.app/Contents/Resources/cura/resources/themes/cura
2015-07-06 20:41:08,494 - INFO - Could not establish connection on /dev/cu.Bluetooth-Incoming-Port: 'Timeout'. Device is not arduino based.
2015-07-06 20:41:09,563 - INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
2015-07-06 20:41:09,689 - DEBUG - Backend connected on port 49674

PyroNyzen, 2015-07-08:
Same error
Mac OSX Early 2015 10.10.3 
Intel Iris Graphics 6100
Cura version 15.06.1 From website.

2015-07-08 15:35:51,387 - INFO - Loaded plugin FileLogger
2015-07-08 15:35:51,616 - INFO - Loaded plugin LocalFileStorage
2015-07-08 15:35:51,623 - INFO - Loaded plugin MeshView
2015-07-08 15:35:51,626 - INFO - Loaded plugin WireframeView
2015-07-08 15:35:51,631 - INFO - Loaded plugin LayerView
2015-07-08 15:35:51,633 - INFO - Loaded plugin OBJReader
2015-07-08 15:35:51,635 - INFO - Loaded plugin STLReader
2015-07-08 15:35:51,637 - INFO - Loaded plugin OBJWriter
2015-07-08 15:35:51,638 - INFO - Loaded plugin STLWriter
2015-07-08 15:35:51,641 - INFO - Loaded plugin GCodeWriter
2015-07-08 15:35:51,644 - INFO - Loaded plugin CameraTool
2015-07-08 15:35:51,649 - INFO - Loaded plugin MirrorTool
2015-07-08 15:35:51,662 - INFO - Loaded plugin RotateTool
2015-07-08 15:35:51,666 - INFO - Loaded plugin ScaleTool
2015-07-08 15:35:51,669 - INFO - Loaded plugin SelectionTool
2015-07-08 15:35:51,672 - INFO - Loaded plugin TranslateTool
2015-07-08 15:35:51,675 - INFO - Checking for new version of cura
2015-07-08 15:35:51,675 - INFO - Loaded plugin UpdateChecker
2015-07-08 15:35:51,688 - INFO - Loaded plugin USBPrinting
2015-07-08 15:35:51,698 - INFO - Could not establish connection on /dev/cu.Bluetooth-Modem: 'Failed to open serial port'. Device is not arduino based.
2015-07-08 15:35:51,700 - INFO - Loaded plugin CuraEngineBackend
2015-07-08 15:35:52,298 - ERROR - Error when loading file .DS_Store: 'utf-8' codec can't decode byte 0xff in position 578: invalid start byte
2015-07-08 15:35:52,473 - DEBUG - Loaded theme /Applications/Cura.app/Contents/Resources/cura/resources/themes/cura
2015-07-08 15:35:53,485 - INFO - Could not establish connection on /dev/cu.Bluetooth-Incoming-Port: 'Timeout'. Device is not arduino based.
2015-07-08 15:35:53,539 - DEBUG - Loaded a mesh with 10452 vertices
2015-07-08 15:35:53,585 - INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
2015-07-08 15:35:53,599 - DEBUG - Backend connected on port 49674
2015-07-08 15:36:07,049 - DEBUG - Loaded a mesh with 4812 vertices

PyroNyzen, 2015-07-08:
It seems to be an error reading the hidden .DS_Store, there is obviously something that python cant decode in UTF-8 format. Is there a reason that DS_Store is being loaded?

awhiemstra, 2015-07-08:
That seems to come from either the theme or available machines list, since those go through a bunch of directories to build a list of files that are used elsewhere in the code. I doubt that is causing the rendering to fail, but if you want, you can try removing the .DS_Store files in the package, they are locatated in Cura.app/Contents/Resources/ and one of the subdirectories, probably cura/resources/themes/cura/icons .

PyroNyzen, 2015-07-08:
The only mention i can find that "might" cause the error is in Cura.app/Contents/Resources/cura/plugins/CuraEngineBackend/Cura_pb2.py Line 477 Referencing default_value=_b("").decode('utf-8')

That's the only mention of the decode function that seems likely to be the cause

redboer, 2015-07-10:
Same error...

fluetke, 2015-07-14:
Same here

Mac OSX Early 2015 10.10.3 
Intel Iris Graphics 6100

redboer, 2015-07-29:
Any updates on this issue?

closed 2015-08-17 by awhiemstra