Ultimaker/Cura #111: Refusal to slice (large model)

CCS86, 2015-07-05:
I am trying to slice a model which is around 90MB. 

Cura 15.06.1 loads the model, displays it correctly, but never begins slicing. The engine log doesn't show any information, but the cura.log file shows this:

Exception in thread Thread-8:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\uranium\UM\JobQueue.py", line 114, in run
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 480, in run
  File "c:\dev\uranium\UM\Mesh\MeshData.py", line 157, in getExtents
  File "c:\dev\python34\lib\site-packages\numpy\lib\arraypad.py", line 1362, in pad
MemoryError

BagelOrb, 2015-07-06:
Sounds like the model is just too large...

nallath, 2015-07-06:
Well, it should still work with very large models.

CCS86, 2015-07-08:
90MB isn't really all that big.

Every other slicer I have tried has chewed right through it. Including 15.04

nallath, 2015-07-08:
Yeah, thats my point. It should work. 90 mb isnt that big.

awhiemstra, 2015-07-08:
I have done some tweaking to make sure we do less data copies. Hopefully that should resolve some of the issues. Unfortunately it turns out Python memory usage is really hard to properly profile so catching every issue is going to be a long and painful process.

nallath, 2015-07-08:
So can this issue be closed?

awhiemstra, 2015-07-08:
Not yet, since I do not know yet if the actual issue is solved.

CCS86, 2015-07-08:
Happy to beta test for you and see

jmrjim, 2015-07-13:
Cura 15.06.1 loads the model, displays it correctly, but never begins slicing. Model size 61MB
Cura 15.02.1 correctly slices the same model. Windows7 64Bit 16MB RAM ASUS.

daid, 2015-07-13:
Model file size does not tell us that much. What we actually want to know is how many triangles are in the model. The previous version of Cura could handle models up to about 3 million of triangles.

awhiemstra, 2015-07-13:
Especially not without the file format, since OBJ or STL will differ in file sizes.

jmrjim, 2015-07-13:
this was 1.246M triangles STL File works fine in older version.
Where is Brim setting on new CURA??

Xisle, 2015-07-13:
[image]

Xisle, 2015-07-13:
More .log info, in case it helps..

Cura 15.06.02 on OS X 10.9.5 just loaded (and displayed after *scaling* for me :| ) a 152.4MB, 1.6M polygon .stl. Then Cura CPU usage dropped and no slicing appeared to be occuring.

I also had Cura 15.04 open having just finished a print - the very one described above. That explains the failure to connect to printer.

`2015-07-13 09:00:11,576 - INFO - Loaded plugin FileLogger
2015-07-13 09:00:12,438 - INFO - Loaded plugin LocalFileStorage
2015-07-13 09:00:12,478 - INFO - Loaded plugin MeshView
2015-07-13 09:00:12,494 - INFO - Loaded plugin WireframeView
2015-07-13 09:00:12,516 - INFO - Loaded plugin LayerView
2015-07-13 09:00:12,528 - INFO - Loaded plugin OBJReader
2015-07-13 09:00:12,536 - INFO - Loaded plugin STLReader
2015-07-13 09:00:12,550 - INFO - Loaded plugin OBJWriter
2015-07-13 09:00:12,562 - INFO - Loaded plugin STLWriter
2015-07-13 09:00:12,582 - INFO - Loaded plugin GCodeWriter
2015-07-13 09:00:12,600 - INFO - Loaded plugin CameraTool
2015-07-13 09:00:12,629 - INFO - Loaded plugin MirrorTool
2015-07-13 09:00:12,749 - INFO - Loaded plugin RotateTool
2015-07-13 09:00:12,768 - INFO - Loaded plugin ScaleTool
2015-07-13 09:00:12,785 - INFO - Loaded plugin SelectionTool
2015-07-13 09:00:12,802 - INFO - Loaded plugin TranslateTool
2015-07-13 09:00:12,822 - INFO - Checking for new version of cura
2015-07-13 09:00:12,823 - INFO - Loaded plugin UpdateChecker
2015-07-13 09:00:12,883 - INFO - Loaded plugin USBPrinting
2015-07-13 09:00:12,913 - DEBUG - Attempting to connect to /dev/cu.usbmodem1a11
2015-07-13 09:00:12,928 - INFO - Loaded plugin CuraEngineBackend
2015-07-13 09:00:14,668 - INFO - Could not establish connection on /dev/cu.usbmodem1a11: 'Timeout'. Device is not arduino based.
2015-07-13 09:00:14,669 - INFO - Could not establish connection on /dev/cu.usbmodem1a11, unknown reasons.
2015-07-13 09:00:17,050 - DEBUG - Loaded theme /Applications/Cura 15.06.02.app/Contents/Resources/cura/resources/themes/cura
2015-07-13 09:00:18,760 - DEBUG - Loaded a mesh with 10452 vertices
2015-07-13 09:00:21,699 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:00:21,712 - DEBUG - Backend connected on port 49675
2015-07-13 09:05:24,770 - DEBUG - Loaded a mesh with 9144354 vertices
2015-07-13 09:06:01,681 - DEBUG - Killing engine process
2015-07-13 09:06:02,218 - INFO - Backend crashed or closed. Restarting...
2015-07-13 09:06:03,431 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:06:03,438 - DEBUG - Backend connected on port 49675
2015-07-13 09:08:24,184 - DEBUG - Killing engine process
2015-07-13 09:08:24,954 - INFO - Backend crashed or closed. Restarting...
2015-07-13 09:08:26,118 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:08:26,132 - DEBUG - Backend connected on port 49675
2015-07-13 09:09:25,417 - DEBUG - Killing engine process
2015-07-13 09:09:25,823 - INFO - Backend crashed or closed. Restarting...
2015-07-13 09:09:31,233 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:09:31,381 - DEBUG - Backend connected on port 49675
2015-07-13 09:09:47,667 - DEBUG - Killing engine process
2015-07-13 09:09:48,191 - INFO - Backend crashed or closed. Restarting...
2015-07-13 09:09:49,501 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:09:57,368 - DEBUG - Backend connected on port 49675
2015-07-13 09:09:59,021 - DEBUG - Killing engine process
2015-07-13 09:09:59,268 - INFO - Backend crashed or closed. Restarting...
2015-07-13 09:09:59,894 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:09:59,975 - DEBUG - Backend connected on port 49675
2015-07-13 09:10:13,335 - DEBUG - Killing engine process
2015-07-13 09:10:13,594 - INFO - Backend crashed or closed. Restarting...
2015-07-13 09:10:14,379 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:10:22,216 - DEBUG - Killing engine process
2015-07-13 09:10:22,311 - DEBUG - Backend connected on port 49675
2015-07-13 09:10:22,423 - INFO - Backend crashed or closed. Restarting...
2015-07-13 09:10:23,042 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-13 09:10:29,797 - DEBUG - Backend connected on port 49675`

CCS86, 2015-07-15:
Engine log in 15.06.2 shows more info:

[libprotobuf ERROR c:/dev/protobuf/src/google/protobuf/io/coded_stream.cc:195] A protocol message was rejected because it was too big (more than 67108864 bytes).  To increase the limit (or to disable these warnings), see CodedInputStream::SetTotalBytesLimit() in google/protobuf/io/coded_stream.h.
Failed to parse message

CCS86, 2015-08-15:
Any headway on this? Still an issue in 15.06.3

jmrjim, 2015-08-17:
I have gone back to earlier version for now.
Regards,
Jim

James N. Greene
JMR Systems Inc.
42 East Derry Road
Derry NH 03038
603-421-1932
www.jmrsystems.com<http://www.jmrsystems.com/>
[JMR Logo MASTER without lower statement transparent]
New England's Resource For Digital Manufacturing

awhiemstra, 2015-08-17:
While I found out how to change the buffer size for Protobuf I have had no time to test whether that resolves the issue. For 15.10 we will also be producing 64-bit Windows builds that hopefully never encounter these issues in the first place.

nallath, 2015-09-23:
Can anyone check if this resolves the issue for them? The beta can be found at http://software.ultimaker.com/Cura_closed_beta/15.10/

closed 2016-12-26 by thopiekar