Ultimaker/Cura #114: Error: Dependency is not satisfiable: python3-pyqt5(>=5.4.0)

ynilsiam, 2015-07-07:
Hello,

I am new to Cura. I download Cura 15.06.01 but when I try to install it gave me this:
Error: Dependency is not satisfiable: python3-pyqt5(>=5.4.0)

I installed PyQt5.4.2. What am I missing?
Thank you.

BagelOrb, 2015-07-07:
What is you operating system?

oliv3r, 2015-07-07:
I think you installed python-pyqt which are the python2 bindings, you will need the Python3 bindings, "python3-pyqt5" for Qt5.

ynilsiam, 2015-07-07:
I am running Linux Mint 17.2.
I got the PyQt5 from http://www.riverbankcomputing.com/software/pyqt/download5
Is this the right one?

Thanks a lot.

ynilsiam, 2015-07-07:
I try this:
sudo apt-get update
sudo apt-get install python3-pyqt5
Anyway, the package is version 5.2.1

so I try this sudo apt-get install python3-pyqt5=5.4.0
sudo apt-get install python3-pyqt5=5.4.2
But both version cannot be found.

How can I get the 5.4.0 version of python3-pyqt5?
Thank you.

oliv3r, 2015-07-07:
Looks like you may be using an old version of linux mint, I'm not aware of their package versions, but I know on debian-jessie (current stable) also only python3-pyqt5-5.2.1 is available. Strech does have 5.4 but is still marked as experimental as distro.

Usually Ubuntu has newer packages and since Mint is a deriviative, they should have too. I could be wrong however.

@ahiemstra told me at the office that we cannot support anything older then 5.4.0 because of a bug in older version of pyqt5. I could however try and see if this bug hits on the 5.2.1 that's on debian however?

ynilsiam, 2015-07-07:
I think I have the latest version of Linux Mint.
I downloaded PyQt5.4.2, then run this code:
python3.4 configure.py
make
make install

However, when I run Cura 15.06.01, it still give me the same error.
How can I check the running PyQt5 version?

Thank you very much.

nallath, 2015-07-07:
Start python3 (type python3 in the terminal)
Then type;
from PyQt5.QtCore import QT_VERSION_STR
print("Qt version:", QT_VERSION_STR)

awhiemstra, 2015-07-07:
Additionally, if you install PyQt manually but want to install the .deb from the site (instead of using the source directly), you will need to skip the dependency checks of dpkg.

nickthetait, 2015-07-07:
Using the `−−ignore−depends` option?

ynilsiam, 2015-07-07:
@nickthetait you mean install Cura by ignore the dependency?
Will that cause other problem when I try to use it?

ynilsiam, 2015-07-07:
Even after install PyQt5.4.2, I still get:
>>> from PyQt5.QtCore import QT_VERSION_STR
>>> print("Qt version:", QT_VERSION_STR)
Qt version: 5.2.1

How can I set the version 5.4.2 to be used?

nallath, 2015-07-07:
Then it seems a path issue. It seems that the place where you installed 5.4.2 is not in the path.

awhiemstra, 2015-07-08:
Actually, the version you get is correct, you are now printing the Qt version. That will still be the system's version of Qt, which in your case is 5.2 . To get the PyQt version, use:

    from PyQt5.QtCore import PYQT_VERSION_STR
    print("PyQt Version:", PYQT_VERSION_STR)

That said, since your system's Qt is 5.2 I do not know if it will work at all. I think we only tested with 5.3 or newer, so you will most likely also need to update your Qt version, making things even trickier.

@nickthetait According to the internet that is the correct switch. I personally would not know for sure though, as I do not use Ubuntu but Arch.

ynilsiam, 2015-07-08:
I installed Qt5.5. I think the problem is the path like @nallath said, but I am not sure how to fix it.
Anyone know how to make Linux see the PyQt5.4.2 that already installed?

daid, 2015-07-10:
I recommend asking the mint community on how to best update these packages, as we can guess and have some linux knowledge. But the mint community will put a lot more eyes on your issue.

ynilsiam, 2015-07-11:
Thank you, @daid I will try that.

nallath, 2015-07-31:
This should be fixed in the latest release.

closed 2015-07-31 by nallath

Thav, 2015-08-17:
For reference, Linux Mint 17.2 is based on the packages from Ubuntu 14.04, Trusty. The first Ubuntu version to have python3-pyqt5 > 5.4.0 is [15.04, Vivid](http://packages.ubuntu.com/search?keywords=python3-pyqt5).

Users of the latest Linux Mint release should be able to pull the packages from 15.04 using [apt pinning](http://askubuntu.com/questions/265111/install-a-single-specific-package-from-future-distribution), but I haven't tried it on my new install yet.

rasterpix, 2015-08-19:
I am running Linux Mint 17.2 and I had the same problem installing Cura and here is what I did to resolve the issue.

1) Add 'deb http://us.archive.ubuntu.com/ubuntu vivid main universe' to your sources.  You can do this by either adding it to your '/etc/apt/sources.list' or you can use the 'Software Sources' app in the GUI.

2) Update your apt cache.  You can do this by launching terminal and running the command 'sudo apt-get update' or you can use the 'Update the cache' button in 'Software Sources' app.

3) Install python3-pyqt5 by launching terminal and running the command 'sudo apt-get install python3-pyqt5'.  

I was then able to install Cura-15.06.03.  Afterword, I did got back and undo the changes that I made in step 1 and then update the cache.  I wanted to make sure that the rest of my packages would not be affected the next time I ran system updates.

Hope this helps.

Tomaszm700, 2015-08-23:
Thanks a lot your solution works fine. In my case I was obligate to install one more packeage from ubuntu repo. sudo apt-get install python3-pyqt5.qtopengl
rest was the same like you do.