Ultimaker/Cura #115: AMF support - materials specified on objects not honored

tamarintech, 2015-07-07:
Hi.  I've started working with a lot of AMF files.  Cura's behavior with multi-material AMF files seems strange to me.  Could someone help clarify the scenarios below?

First scenario:
My AMF file contains one object with two volumes.  The volumes may or may not contain a material ID.

The expected result is that the volumes would all be printed with the default material.  Actual result is the second volume always gets designed for print by the second extruder.

Second Scenario:
My AMF file contains two separate objects that define one volume per each object.  Each volume specifies a separate material ID (1 and 2).

The expected result is that each object would be imported into the Cura scene and the defined volumes with separate material IDs would be printed with extruder 1 and extruder 2, respectively.  The actual result is that both objects are printed with extruder 1.

nallath, 2015-07-07:
AMF is not supported for 15.06 yet. Are you using 15.04?

tamarintech, 2015-07-07:
I was testing in 14.

nallath, 2015-07-07:
14 what? 14.x? That is a -really- old version.

tamarintech, 2015-07-07:
14.12.1 from the Fedora 22 repositories. ;)

quillford, 2015-07-07:
@tamarintech wrong repo. you are using an old version from [daid/Cura](https://github.com/daid/Cura).

quillford, 2015-07-07:
@tamarintech see [this](https://github.com/daid/Cura/blob/9b5aa5c182a6c2285f1f5f362ac0eeb31d3ff222/Cura/util/meshLoaders/amf.py)

tamarintech, 2015-07-07:
@quillford Understood.  I am not requesting the previous version to be fixed, but instead requesting the behavior be evaluated under the assumption that the previous support for AMF would be re-implemented similarly.

Would it be more beneficial to submit patches to the old branch regarding AMF loading?  I can close this and work on that instead, unless AMF support is already being re-implemented elsewhere.

nallath, 2015-07-07:
We will not develop new features for the old Cura and expect that this version will eventually completely replaces the old one.

closed 2015-07-07 by nallath

awhiemstra, 2015-07-08:
However, having a discussion on the intended behaviour of the AMF support is a good thing. Of course, what I have understood from @daid is that the support for AMF in Cura is based on a leaked draft of the AMF specification. I do not know if the final has been fixed yet and if so, if any of your points have been addressed in that.

> Would it be more beneficial to submit patches to the old branch regarding AMF loading? I can close this and work on that instead, unless AMF support is already being re-implemented elsewhere.

As nallath mentioned, we won't develop new features for the old branch. Currently there is no AMF file loader plugin for the new branch, but both the OBJ and STL reader plugins are almost literally the code from the old branch, so should not be too hard to port over.

One thing to keep in mind however is that both dual extrusion features and per-object settings which are required for correct handling of these situations are not yet implemented. They are planned but this issue will not really be resolvable until those features are implemented.

daid, 2015-07-08:
The AMF spec does not really describe how something should be interpreted for printing. It only says that you can define objects, and have materials on volumes, and a few more places.

The current legacy Cura AMF implementation is only a partial implementation. It also does not support the curved triangles.

The above problem with multiple objects with different material settings per object is also an legacy Cura limitation. Due to how multi material printing is implemented there.

daid, 2015-07-08:
(Also, it would be good to know which applications can export AMF now. As when the implementation in Cura was made, there where no applications that could do it. Except for the example implementation, which could exports files that did not match the specs)

tamarintech, 2015-07-08:
Thanks for continuing the discussion.

Conversion from popular formats to AMF (basic support only) is relatively trivial and supported by most popular printing software suites.  While the standard is not free we do have some implementation specifics such as the XSD here - http://amf.wikispaces.com/file/view/amf.xsd/402895728/amf.xsd and current workflows already using limited-scope AMF.  AMF has certainly performed well as a bandaid to the problem of multi-tool and multiple model scene compilation over the past years.

At this time, AMF is the only standard implementation we have for supporting dual extrusion, model archives and rich metadata.  3MF is quickly on its way, but currently has little community support (or enthusiasm).  A completely new printing file format would be very difficult to gain traction with.  These are the primary reasons I have been involving AMF in my workflow lately.  What other options do we have or would there be interest in?

nallath, 2015-07-09:
Well, any option would be good. It's not that hard to add new file formats to Cura.