Ultimaker/Cura #118: Some fine tuning on scrolling through layers in layer view mode

Fass-ohne-Hoden, 2015-07-07:
In layer view mode, you can scroll through layers with your mouse wheel (or whatever mimics scrolling on your machine). Scrolling through layers (instead of zooming in/out) is enabled when the cursor is placed on the slider bar.

1) Enlarge the zone for the abovementioned scrolling behaviour a little bit. Make that zone visible (e.g. by adding a visible background to the slider or a frame around it).

2) Make the scroll rate adjustable (with one layer per "scroll increment" as lowest setting). Have a default setting where the scroll rate is taken from the OS' settings ...
The latter one is not really required, just a nice to have kind of thing.

awhiemstra, 2015-07-08:
This is actually default behaviour of the slider control we use, so point 2 is going to be hard to do without reimplementing the control. Point 1 is a good one I have also seen, we will look into that.

nallath, 2015-07-08:
We could fake it by using a slider that goes from 1 <-> ((max layers) / steps). In the normal case steps would be 1, but for faster scrolling you could increase it.

Fass-ohne-Hoden, 2015-07-08:
@ awhiemstra
Which is default behaviour?
If it matters, my OS' setting is to scroll one line per increment. In Cura, with a 600+ layer print, the view jumps, like,  a dozen or more layers per scroll step.

If I may add this -- I just found out that, sporadically, scrolling moves the slider AND zooms at the same time. If you could have a look at that, too, please :)

closed 2015-07-09 by TMHogenhout

Fass-ohne-Hoden, 2015-08-17:
I have no idea what that code means.

I take it you won't implement a means to scroll only one layer per (wheel) step?

Ghostkeeper, 2015-08-18:
It looks like 1) is implemented but 2) isn't, and nothing in the code seems to be aimed at the additional bug you found that scrolling would also zoom.

TMHogenhout, 2015-08-19:
Maybe some explanation is required. When your mouse is on the build plate area; the scroll wheel triggers the zoom-function. But when your mouse is on the layer view slider area; the scroll wheel steps through the layers. In the past, the layer view slider area was very small (what made it trigger the zoom function when your pointer was displaced even a little bit) and the actual area was not visible. So I made the area where you can use the scroll wheel to step through the layers, a lot bigger and visible. 

I have not implement a way to make the scroll rate adjustable, for the very sheer reason that the layer view scroll bar will be redesigned and will perform different behaviour. It will probably have a little input field on the handle that will allow you to type in the precise layer that you want to view. When this input field is selected you can also step through the layers one step at a time with the arrow buttons. But I will pick this up with our designer to make a definitive design for this behaviour.

commit ae89ef37 (2015-07-09): Enlarge the zone where you can use scrolling to look trough the layer mode.

Also make it more visible

fixes #118