Ultimaker/Cura #119: Make width of the sttings bar adjustable

Fass-ohne-Hoden, 2015-07-07:
The new Cura has settings bar on the right side, with a fixed width. In some languages, a lot of the settings captions are shortened to unreadability.

Make the width of that settings bar adjustable (in a grab-the-border-and-drag-it-around kind of way).

nallath, 2015-07-07:
Would changing a preference setting be sufficient?

Fass-ohne-Hoden, 2015-07-07:
You mean, like, inputting the width in pixels?
... maybe not the most intuitive option, but it would work, sure. In that case, could it be made so that the setting takes effect instantly (at least, without restarting the program)? Although, one is likely to configure this only once or twice, it'd be a hassle to iterate through that thing (change value; restart; check [crap, too much]; changing value again, restart ...)

nallath, 2015-07-07:
Yeah I agree, but I'm not quite sure how hard it would be to make it dragable. Hence my question.

awhiemstra, 2015-07-08:
Draggable should be doable, but I would rather make sure the default width of the sidebar is correct. Of course, that is quite a bit harder to do.

The primary reason for this is that I want to make it possible to put the sidebar in a two or more column mode so you can have a better overview of all the settings available on larger screens. This would be someting I think makes sense to have attached to a draggable edge of the sidebar.

BagelOrb, 2015-07-08:
I think this feature is also nice to have when you have a small screen. 

Putting the settings into two (or more) bars wouldn't necessarily create more overview. You should then always cut off the settings at [Travel], instead of an adjustable cut. That way the placement of the settings is more predictable.

I think this issue is low priority, though.

nallath, 2015-08-14:
Layout will be changed, see https://github.com/Ultimaker/Cura/issues/194

closed 2015-08-14 by nallath