Ultimaker/Cura #121: Unable to install Cura 15.06.1 on Ubuntu 15.04

WebfinityDesign, 2015-07-07:
I'm unable to install Cura 15.06.1 on Ubuntu 15.04.

On install I get the message:

Preparing to unpack /tmp/[url="Cura-15.06.01-Linux.deb"]Cura-15.06.01-Linux.deb[/url] ...
Unpacking cura ([url="15.06.01"]15.06.01[/url]) over ([url="15.04"]15.04[/url]) ...
dpkg: error processing archive /tmp/[url="Cura-15.06.01-Linux.deb"]Cura-15.06.01-Linux.deb[/url] (--install):
 trying to overwrite '/usr/bin/protoc', which is also in package protobuf-compiler [url="2.6.1-1"]2.6.1-1[/url]
dpkg-deb (subprocess): decompressing archive member: internal gzip write error: Broken pipe
dpkg-deb: error: subprocess <decompress> returned error exit status 2
dpkg-deb (subprocess): cannot copy archive member from '/tmp/[url="Cura-15.06.01-Linux.deb"]Cura-15.06.01-Linux.deb[/url]' to decompressor pipe: failed to write (Broken pipe)

The upgrade to Cura 15.04 went without problems.
Any idea what to do?

Patola, 2015-07-08:
You have to remove protobuf-compiler and libprotobuf-dev otherwise it won't install.

awhiemstra, 2015-07-08:
Unfortunately the 15.06.01 package includes some protobuf files that conflict with existing packages. I have already fixed this in the packaging scripts and the next version will not have that issue. In the meantime, you can uninstall protobuf-compiler and libprotobuf-dev as mentioned.

closed 2015-07-08 by awhiemstra

WebfinityDesign, 2015-07-08:
Ok. That did the trick. thanks