Ultimaker/Cura #122: Cannot Update Firmware

Fuzionist1, 2015-07-08:
I cannot updating the firmware. I have a new UM2.  I have never updated the firmware before. Windows 7. This is what I did:

1. Installed Cura 15.06.01
2. Plugged in UM2
3. Computer said that device not configured - couldn't find drivers.
4. Downloaded and installed Arduino drivers
5. Ultimaker2 showing as Arduino Mega 2560 (COM4) in Device Manager
6. Opened Cura as administrator
7. Selected Update Firmware
8. Progress bar doesn't move at all (waited 10+ minutes)
9. Cancelled the update.
10. Opened Cura.log, which reads as follows:

Exception in thread Thread-6:
Traceback (most recent call last):
File "C:\Program Files (x86)\Cura_15.06.01\plugins\USBPrinting\avr_isp\stk500v2.py", line 29, in connect
self.serial = Serial(str(port), speed, timeout=1, writeTimeout=10000)
File "c:\dev\python34\lib\site-packages\serial\serialwin32.py", line 38, in __init__
File "c:\dev\python34\lib\site-packages\serial\serialutil.py", line 282, in __init__
File "c:\dev\python34\lib\site-packages\serial\serialwin32.py", line 66, in open
serial.serialutil.SerialException: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
File "c:\dev\python34\lib\threading.py", line 869, in run
File "C:\Program Files (x86)\Cura_15.06.01\plugins\USBPrinting\PrinterConnection.py", line 140, in _updateFirmware
programmer.connect(self._serial_port)
File "C:\Program Files (x86)\Cura_15.06.01\plugins\USBPrinting\avr_isp\stk500v2.py", line 31, in connect
raise ispBase.IspError("Failed to open serial port")
USBPrinting.avr_isp.ispBase.IspError: 'Failed to open serial port'

awhiemstra, 2015-07-10:
I am currently working on several issues related to USB printing. This should hopefully all be resolved for the next patch release.

daid, 2015-07-10:
Note, new UM2's should come with the latest firmware installed already (15.04 at this time of writing)

Fuzionist1, 2015-07-10:
Good to know.

closed 2015-08-14 by nallath