Ultimaker/Cura #123: Cura 15.06.01: Layer view missing visualisation of top layers if concentric pattern is selected

Spirou42, 2015-07-08:
Steps to Reproduce
1. have a simple cube in cura
2. enable top/bottom pattern in the machine settings.
3. set the top/bottom pattern to concentric
4. enable layers view
Result:
[image]
 expected result:
something like 
[image]

nickthetait, 2015-07-08:
I may be wrong, but concentric might deliberately disable the top surface. So it would be expected that you would not see/print that part.

Spirou42, 2015-07-08:
I have exported the GCode and rendered it with Repetier-Host. The top and bottom layers are complete, including a nice concentric pattern.
[image]
[image]

nickthetait, 2015-07-08:
I was definitely wrong! That's what I get for trying to help :disappointed:

nallath, 2015-07-09:
I didn't even knew we had a concentric pattern :$

luisonoff, 2015-07-22:
I confirm this is also happening to me. With concentric pattern chosen, no top layers show up on layer view.

nallath, 2015-07-23:
Is this still an issue with 15.06.02?

Xisle, 2015-07-23:
Yes.

[image]

Xisle, 2015-07-23:
No top or bottom.

[image]

BagelOrb, 2015-08-25:
Visualizing the concentric pattern should be fairly easy.

However, visualizing ZigZag infill will probably be harder.

nallath, 2015-09-23:
Currently no skin is rendered, it seems like it's not correctly sent by the engine.

BagelOrb, 2015-09-29:
This should be fixed for the next release.
I now do get skin visualization for any pattern.

closed 2015-09-29 by BagelOrb