Ultimaker/Cura #125: Cura 15.06.01crashes

3dpc, 2015-07-08:
Each model I am trying to slice (upload to tray) cause CuraEngine to crash
Be happy to supply more info on this subject, where is the log file?

nallath, 2015-07-09:
Depends on your operating system. Which one do you use?

3dpc, 2015-07-09:
Windows 7

nallath, 2015-07-09:
You should run it in admin mode. Windows 7 doesn't always let software write to file otherwise. If I recall correctly it should create a *.log file in the exectuable directory.

3dpc, 2015-07-09:
Now it does not interact at all, yesterday it allowed me to open and load
stl file, now it just dont interact
Can choose any menu.
I tried in admin and in regular mode, tried unistalling and installing but
nothing help

3dpc, 2015-07-09:
Restarted my computer, Cura is interacting again but still crashes (also in
admin mode)

nallath, 2015-07-09:
Did it create a log file? It should be named Cura.log

3dpc, 2015-07-09:
Yes it did,
Here are the last lines in it, not much info..

2015-07-09 11:15:12,917 - INFO - Checking for new version of cura
2015-07-09 11:15:12,917 - INFO - Loaded plugin UpdateChecker
2015-07-09 11:15:12,930 - INFO - Loaded plugin USBPrinting
2015-07-09 11:15:12,932 - INFO - Loaded plugin CuraEngineBackend
2015-07-09 11:15:13,655 - DEBUG - Loaded theme C:/Program Files
(x86)/Cura_15.06.01/resources/themes/cura
2015-07-09 11:15:14,148 - INFO - Started engine process: C:\Program Files
(x86)\Cura_15.06.01\CuraEngine.exe
2015-07-09 11:15:14,457 - DEBUG - Backend connected on port 49674
2015-07-09 11:15:18,249 - DEBUG - Loaded a mesh with 31698 vertices
2015-07-09 11:15:19,455 - DEBUG - Killing engine process
2015-07-09 11:15:19,459 - INFO - Backend crashed or closed. Restarting...
2015-07-09 11:15:19,714 - INFO - Started engine process: C:\Program Files
(x86)\Cura_15.06.01\CuraEngine.exe
2015-07-09 11:15:19,761 - DEBUG - Backend connected on port 49674
2015-07-09 11:15:27,781 - INFO - Backend crashed or closed. Restarting...
2015-07-09 11:15:28,041 - INFO - Started engine process: C:\Program Files
(x86)\Cura_15.06.01\CuraEngine.exe
2015-07-09 11:15:28,117 - DEBUG - Backend connected on port 49674
2015-07-09 11:31:43,242 - DEBUG - Killing engine process
2015-07-09 11:31:43,246 - INFO - Backend crashed or closed. Restarting...
2015-07-09 11:31:43,501 - INFO - Started engine process: C:\Program Files
(x86)\Cura_15.06.01\CuraEngine.exe
2015-07-09 11:31:43,557 - DEBUG - Backend connected on port 49674
2015-07-09 11:31:48,221 - DEBUG - Loaded a mesh with 31698 vertices
2015-07-09 11:31:49,373 - DEBUG - Killing engine process
2015-07-09 11:31:49,377 - INFO - Backend crashed or closed. Restarting...
2015-07-09 11:31:49,633 - INFO - Started engine process: C:\Program Files
(x86)\Cura_15.06.01\CuraEngine.exe
2015-07-09 11:31:49,680 - DEBUG - Backend connected on port 49674
2015-07-09 11:31:55,933 - INFO - Backend crashed or closed. Restarting...
2015-07-09 11:31:56,198 - INFO - Started engine process: C:\Program Files
(x86)\Cura_15.06.01\CuraEngine.exe
2015-07-09 11:31:56,274 - DEBUG - Backend connected on port 49674

nallath, 2015-07-09:
Hmm. Yeah, nothing there that indicates what is going wrong. Could you try to run Cura from console and report what it says? Might be that there are errors that are not caught by the logging.

3dpc, 2015-07-09:
well here is what I managed to get in the log file:
Exception in thread Thread-5:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\uranium\UM\JobQueue.py", line 114, in run
  File "c:\dev\uranium\UM\Mesh\ReadMeshJob.py", line 38, in run
AttributeError: 'NoneType' object has no attribute 'getExtents'

Exception in thread Thread-5:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\uranium\UM\JobQueue.py", line 114, in run
  File "c:\dev\uranium\UM\Mesh\ReadMeshJob.py", line 38, in run
AttributeError: 'NoneType' object has no attribute 'getExtents'

nallath, 2015-07-09:
Ah! That's something we can work with. Thanks.

3dpc, 2015-07-09:
You are welcome, please update when you have a new version that solve this
issue.
I'll be happy to test it and confirm if issue has solved (and get the
possibility to work with the new version)

Assaf

BagelOrb, 2015-07-09:
What version of Cura are you running?

3dpc, 2015-07-09:
15.06.01

nallath, 2015-07-31:
Is this resolved in 15.06.03?

3dpc, 2015-08-01:
I dont experience the crushes any more.
Thanks,
Assaf
http://www.3dprintingcenter.co.il/

closed 2015-08-03 by awhiemstra

Dim3nsioneer, 2015-08-03:
I had a start crash with Cura 15.06.3 on a Win8.1 machine. Checking the Windows crash log revealed a crash within the graphic drivers. Updating the drivers solved the problem.