Ultimaker/Cura #126: Wire Print Crashes shortly after being selected from options

Tom-Lauerman, 2015-07-08:
I'm excited about the wire print function. However, at present it seems to cause Cura to crash almost immediately after selecting wire print. Several people on the UM forum have found this same problem.

nallath, 2015-07-09:
@BagelOrb Could you have a look at this?

BagelOrb, 2015-07-09:
Thanks for communicating this to us. I will dig into it.

BagelOrb, 2015-07-09:
When I slice using the engine directly I don't encounter any problem.

Supposing that the defaults correspond to the inherited values enough, this is most probably a frontend issue.

Perhaps it has something to do with the fact that the layer height communicated to the frontend is around 1 cm..

@awhiemstra can you take a look at it?

BagelOrb, 2015-07-09:
Fixed!

closed 2015-07-09 by BagelOrb

BagelOrb, 2015-07-09:
Can you provide a link to the forum post?

Tom-Lauerman, 2015-07-09:
Sure, here you go:
https://ultimaker.com/en/community/view/16675-has-anyone-had-any-luck-with-wireprint

Thanks!