Ultimaker/Cura #127: Cura_15.06.01 Crash at its startup

s-marashi, 2015-07-09:
As I try to run it on Windows 7 32Bit it crash at its first time startup it shows splash window then master window and add printer window is shown but immediately after that it crash which I can't select my printer type and windows application crash pops up and my only selection is to close program.
As you mentioned in #125 I tried to run in admin mode and from command line  but everything goes in the same way.

nallath, 2015-07-09:
Then isn't this the same issue?

s-marashi, 2015-07-09:
No, as in #125 it crash when he wants to slice layers but in this one it crashes at application startup..

nallath, 2015-07-09:
Allright, could you post the log file?

s-marashi, 2015-07-09:
I expected it creates one when I run it as admin but it didn't so there is nothing to send.

nallath, 2015-07-09:
Could you run it in command prompt and copy paste what displays there?

s-marashi, 2015-07-09:
Again it didn't produce any thing.

nallath, 2015-07-09:
So you start the software and it does nothing? Not even when running it from console?

s-marashi, 2015-07-09:
Yes, It just shows splash , master and add printer window then crash.

BagelOrb, 2015-07-09:
Did you try running Cura as administrator?

nallath, 2015-07-09:
Could you check in user/appdata/local/cura? Should be a log file there.

s-marashi, 2015-07-09:
Yes, I run it as admin from both explorer and command prompt.

s-marashi, 2015-07-09:
Logs was there:
2015-07-09 17:58:42,361 - INFO - Loaded plugin FileLogger
2015-07-09 17:58:42,648 - INFO - Loaded plugin LocalFileStorage
2015-07-09 17:58:42,661 - INFO - Loaded plugin LayerView
2015-07-09 17:58:42,675 - INFO - Loaded plugin MeshView
2015-07-09 17:58:42,689 - INFO - Loaded plugin WireframeView
2015-07-09 17:58:42,697 - INFO - Loaded plugin OBJReader
2015-07-09 17:58:42,702 - INFO - Loaded plugin STLReader
2015-07-09 17:58:42,711 - INFO - Loaded plugin STLWriter
2015-07-09 17:58:42,717 - INFO - Loaded plugin GCodeWriter
2015-07-09 17:58:42,730 - INFO - Loaded plugin CameraTool
2015-07-09 17:58:42,743 - INFO - Loaded plugin MirrorTool
2015-07-09 17:58:42,784 - INFO - Loaded plugin RotateTool
2015-07-09 17:58:42,799 - INFO - Loaded plugin ScaleTool
2015-07-09 17:58:42,811 - INFO - Loaded plugin SelectionTool
2015-07-09 17:58:42,825 - INFO - Loaded plugin TranslateTool
2015-07-09 17:58:42,844 - INFO - Checking for new version of cura
2015-07-09 17:58:42,845 - INFO - Loaded plugin UpdateChecker
2015-07-09 17:58:42,872 - INFO - Loaded plugin USBPrinting
2015-07-09 17:58:42,877 - INFO - Loaded plugin CuraEngineBackend
2015-07-09 17:58:43,783 - INFO - Could not establish connection on COM3: 'Failed to enter programming mode'. Device is not arduino based.
2015-07-09 17:58:44,228 - DEBUG - Loaded theme D:/Cura_15.06.01/resources/themes/cura
2015-07-09 17:58:44,664 - INFO - Started engine process: D:\Cura_15.06.01\CuraEngine.exe

s-marashi, 2015-07-09:
Now, do you have any idea how should I launch it or it is a bug?

nallath, 2015-07-09:
I don't see anything that could give an indication as to why it crashes.

s-marashi, 2015-07-09:
Does it open serial ports just on startup?
As it said "Could not establish connection on COM3"
Or it just have problem with arduino IDE?
Excuse me but I really want to run it.

nallath, 2015-07-09:
That's the USBPrinting plugin that tries to open all USB devices to check if they are 3D printers. It can't open it, so its probably not a printer.
I might have one idea why it doesn't work; Could you post your driver / video card info?

s-marashi, 2015-07-09:
But nothing is not connected to USB ports.
Where should I search for video card info?

s-marashi, 2015-07-09:
I think it should be:"Mobile Intel(R) 4 Series Express Chipset Family"

awhiemstra, 2015-07-09:
This sounds like a duplicate of #96

s-marashi, 2015-07-09:
I think so, now do I should close this issue and follow #96?

nallath, 2015-07-09:
Yes

closed 2015-07-09 by nallath