Ultimaker/Cura #129: Allow comma as decimal mark (again)

Fass-ohne-Hoden, 2015-07-09:
Cura 15.06 only accepts a dot as decimal mark (or did I miss a setting?)
This is annoying if your Numpad layout has a comma (as, perhaps, you live in a country where the comma is THE decimal mark)

The older Cura accepted both comma and dot as decimal mark. Which is as it should be (unless both were required for different purposes, which isn't the case here). Make the new Cura the same.

BagelOrb, 2015-07-10:
You request is valid, but currently low priority. There are a lot of pressing issues at the moment.

nallath, 2015-07-13:
This has something to do with your locale settings I believe. Your operating system decides what can be used as a decimal point.

awhiemstra, 2015-07-13:
Actually, in this case it sounds more like a keyboard layout thing. The locale issues were solved by enforcing using . everywhere. Since we now enforce using . everywhere, a keyboard that uses , instead of . on the numpad will be a bit more annoying to use.

Fass-ohne-Hoden, 2015-07-13:
It's neither an issue with locale settings nor keyboard layout - both are just right the way they are. It is a program issue if the program that is aimed at and used in numerous countries doesn't adopt (either by fully accepting both comma and dot as decimal mark or by converting one into the other internally). However, I do understand that there are more pressing issues right now and some things need to wait.

MarcusWolschon, 2016-05-01:
if texts are translated into the users locale (in 2.1beta I can select a language), 
then the formatting of dates, times and numbers of that chosen locale or in the absence thereof of system locale of the current used must be used. 
If my locale uses a comma for decimal point and a dot as thousand separator, then that is how all numbers must be entered and displayed.

BagelOrb, 2016-05-01:
I don't agree. My locale suggest using the cons as decimal separator, but
I'm used to using dots. I think handling both as decimal separator is more
permissive towards users.

I don't see much use for a thousand separator in Cura; values generally
don't go that high.

daid, 2016-05-02:
The old code allowed math as well as commas, for reference:
https://github.com/daid/Cura/blob/SteamEngine/Cura/util/profile.py#L913

nallath, 2016-05-02:
The new one simply uses dots for decimals and forces everything to that. So it doesn't matter what you enter, the result is the same.

closed 2016-05-02 by nallath

daid, 2016-05-02:
I noticed with older Betas the behaviour is a bit undefined if you fill in commas. But the open beta seems to behave as Nallath describes. (It replaces any entered comma with a dot when you enter it)

nallath, 2016-05-02:
A few of the closed beta testers also noticed this behavior, so it was fixed at some point.