Ultimaker/Cura #130: Cura cannot be uninstalled completely.

eelcor, 2015-07-09:
Issue affects both Mac and Windows. My windows machine gives an error when starting Cura. This error is the result of a previous crash and is somehow not reset when uninstalling and reïnstalling Cura.

For the Mac I've added a Prusa i3 settings file and somehow it is not found. After reinstalling and trying without the settings file, Cura remembers the earlier settings and throws an error.

This is a pity, because the print I was able to make was really nice!

eelcor, 2015-07-09:
Okay the Mac issue has been solved by removing the .Cura file from my home dir.

awhiemstra, 2015-07-10:
We should probably add an option to the windows uninstaller to remove existing settings. Unfortunately this is going to be hard for the OSX installer since I do not know if it has uninstall hooks.

eelcor, 2015-07-10:
For OSX it suffices to say that you can remove the .Cura directory. For Windows the settings are less transparent. Where can I find these?

awhiemstra, 2015-07-10:
~/AppData/Local/Cura

quillford, 2015-07-10:
Mac was an issue for me as well, but you just need to make another json file with the same one as you were using before.

BagelOrb, 2017-07-21:
It would still be nice to have an option in the deinstaller to delete all configuration files.

Ghostkeeper, 2017-10-30:
Our project manager just removed this from our planner because it's older than 12 weeks.

ianpaschal, 2018-05-22:
@Vandrasc Can you put this ticket number in JIRA and add the JIRA number here?

Vandrasc, 2018-05-23:
Yep, sure can! See CURA-5154.

diegopradogesto, 2018-07-13:
This is now implemented in the master branch.

closed 2018-07-13 by diegopradogesto