Ultimaker/Cura #132: CuraEngine has stopped working - on large files

PatrickHu2e, 2015-07-10:
Version 15.06 - (nice improvement but needs work)...running on Windows 8.1
I need to use the Ultimaker 2 Extended for its size and quality, however I can not prepare my files on this new version of Cura - the CuraEngine fails when calculating the time for "Maximum Quality".  One of my test STL files is 64MB,  the object has a volume of 367 ccm,, area 1812 ccm, 1,319,476 triangles.  Its dimensions are 100mm X 150mm X 230mm (note: older versions would warn me it was too large to print on its side, this version does not).  It can complete all calculations on much smaller objects (even multiples), but not large ones.  On the large ones, it can complete the average and "Minimum Draft", but crashes before it can complete the "Maximum Quality", even when I set the slider to maximum.  The error log provided the following from two attempts:
 
Exception in thread Thread-18:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\libArcus\python\Arcus\__init__.py", line 228, in run
  File "c:\dev\libArcus\python\Arcus\__init__.py", line 295, in _receiveNextMessage
  File "c:\dev\libArcus\python\Arcus\__init__.py", line 337, in _receiveBytes
MemoryError

Exception in thread Thread-10:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\uranium\UM\JobQueue.py", line 114, in run
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 480, in run
  File "c:\dev\uranium\UM\Mesh\MeshData.py", line 157, in getExtents
  File "c:\dev\python34\lib\site-packages\numpy\lib\arraypad.py", line 1370, in pad
  File "c:\dev\python34\lib\site-packages\numpy\lib\arraypad.py", line 143, in _append_const
MemoryError

Will we ever see a 64-bit version of Cura?  It would help.

daid, 2015-07-10:
Yes, 64bit will solve this issue quite well. We're planning to go to an improve packaging system for windows soon which should help in building the 64bit version. (Linux and Mac are already 64bit by default right now)

Hinsow, 2015-07-11:
I am having the same issus. I cannot print one of my large parts. Here is the Engine log;

[image]

Then it just frizzes, iv tried this about 5-6 time. I current using the 64 Bit - 15.06 version.

Any help would be great as i only brought this print mainly for this part 4 days ago... CHEERS

daid, 2015-07-11:
(There is no 64bit windows version right now, so you cannot be using that one)

Also note that the old version of Cura is available at:
http://software.ultimaker.com/current/

Which might work better if you are on windows and need to prepare these huge files in the meantime.

awhiemstra, 2015-07-13:
We should have a build out soon that hopefully improves memory usage. Additionally, as @daid mentioned, we will have a 64-bit build for the 15.10 release.

closed 2015-08-14 by nallath