Ultimaker/Cura #133: Cura cannot write to Cura.log in Program Files

fieldOfView, 2015-07-10:
When installed in the Program Files (x86) folder, Cura can not write to Cura.log (unless Cura is started in elevated mode). So when a Python error occurs, it is impossible to figure out what went wrong. Is there any way this file could be created at some other location? (Somewhere in AppData, or in the .cura folder in the profile folder which legacy Cura used)

Workarounds:
* Start Cura in elevated mode ("as Administrator")
* Install Cura to a location outside %ProgramFiles%

Fix:
* Store Cura.log in %AppData%/Cura (prefered) or %UserProfile%/.cura

nallath, 2015-07-13:
Yeah, agreed. We should change the default of where the log is placed.

awhiemstra, 2015-07-13:
The actual Cura log is already stored in %AppData%/Local/Cura . The error message is from Py2Exe which should have code to deal with not being able to write but apparently does not handle it correctly.

commit 8130bae4 (2015-07-13): Add a crash handler to catch uncaught exceptions

This should catch any uncaught exceptions and avoid the Py2Exe message
about Cura.log

Fixes #133
closed 2015-07-14 by awhiemstra