Ultimaker/Cura #134: Scrolling through Advanced settings can alter drop-down settings

Xisle, 2015-07-11:
Application Version: 15.06.02
Platform: OS X 10.9.5

Steps to Reproduce:
Switch to Advanced settings
Position mouse cursor vertically over any drop-down menu(s)
Scroll through settings using mouse wheel

Actual Results:
Drop-down menu selections change as they are scrolled over

Expected results:
No settings would change while scrolling through Advanced settings 

I would be fine with the drop-downs not responding to mouse scroll events at all.

salscode, 2015-07-13:
+1 Experienced the same issue.

App: 15.06.01
Platform: Windows 8.1

BagelOrb, 2015-07-15:
Note that while the active enum setting changes by scrolling, the blue highlighted enum setting in the dropdown doesn't change along. Either deselect/unhighlight all enum settings in the dropdown, or highlight the currently selected one.

awhiemstra, 2015-10-27:
This should be fixed in the most recent beta.

closed 2015-10-27 by awhiemstra