Ultimaker/Cura #135: CuraEngine.exe crashes on Windows10

elettrozero, 2015-07-11:
Stacktrace from Windows Event Viewer:
Faulting application name: CuraEngine.exe, version: 0.0.0.0, time stamp: 0x5592b7e5
Faulting module name: libgcc_s_dw2-1.dll, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc0000094
Fault offset: 0x00006f6b
Faulting process id: 0x1168
Faulting application start time: 0x01d0bc192f713de3
Faulting application path: C:\Cura_15.06.01\CuraEngine.exe
Faulting module path: C:\Cura_15.06.01\libgcc_s_dw2-1.dll
Report Id: f8bb9f30-e7f7-4e0b-abb8-7a70c9554a69
Faulting package full name: 
Faulting package-relative application ID: 

Already tried every compatibility settings and running the program as Administrator: nothing worked.

nallath, 2015-07-13:
Could you post your graphic card info and log file?

awhiemstra, 2015-07-13:
This sounds like an incompatibility between Windows 10 and libgcc. Unfortunately I do not know the status of MinGW for Windows 10. You will probably need to wait until at least the final release before there is any update on that.

ThatJCB, 2015-07-25:
I get the same symptoms on Windows 8.1.
Is there a separate thread for that?

daid, 2015-07-25:
There has been a general crash bug identified a while back. @BagelOrb isn't this bug just the result of that crash?

elettrozero, 2015-07-25:
Cura and Cura engine are working fine with latest update on latest version of windows 10.

closed 2015-08-10 by awhiemstra