Ultimaker/Cura #142: Too large items are scaled to print volume, but lose original dimensions

Ghostkeeper, 2015-07-14:
Version: 15.06.01
Platform: Windows 7

Steps to reproduce:
* Find a model that is too large for the print volume, such as [this one](http://dulek.net/persoonlijk/LargeSphere.stl). Load it into Cura. Take a look at the scaling menu.

Actual results:
* The model is scaled to fit within the print volume, but the reported scale is 100%. This way there is no way to actually get the original dimensions of the item back.
* Setting a different scale, such as 50%, results in a print that fills 50% of the dimensions of the build volume rather than being 50% of the actual size of the 3D model.

Expected results:
* The model is scaled to fit within the print volume, and Cura reports the scale it now has (e.g. 20% if it had to be scaled by a factor of 5 to fit within the print volume).
* Setting a different scale should resize the item to the specified scale with respect to the item's original scale, regardless of the print volume.

Additional information:
* The model I linked to in this bug report should be a (tesselated) sphere with a radius of 10m. It was created with OpenSCAD with the code "sphere(r=10000);"
* This issue was originally reported by Niyoki [on the Ultimaker forum](http://ultimaker.com/en/community/view/16737-stl-object-dimensions-size-1504-vs-150602).

BagelOrb, 2015-07-14:
Sometimes the model is specified in meters instead of millimeters. You then don't want scales of 1%. 1 meter should then correspond to 1 cm or something.

Perhaps when applied scaling is less than 1%, multiply the displayed scaling by 100, so that the displayed scale is always between 1% and 999%.

It might also be an idea to scale up if the model is too small... For example when all dimensions are smaller than 1mm.

BagelOrb, 2015-07-14:
Also it would be better if scaling only occurs in steps of ten. So scale by 10^x, instead of scale to fit.
(See https://github.com/Ultimaker/Cura/issues/14)

Ghostkeeper, 2015-07-14:
After scaling the model in the example, 1 unit length equals 0.01025mm (so it's scaled to 1.025%). It is then logical to assume that the units of this model were in 100ths of millimetres, or half thous.

In Niyoki's original problem though, by the looks of it the model was scaled to approximately 60%. That should be undesired behaviour, I think.

nickthetait, 2015-07-14:
:+1: Don't forget to give a little notification. Something like: "Model too large, scaled down by X%"

Xisle, 2015-07-14:
I lean toward NO auto scaling. A model that is too large does not always need uniform scaling. How about popping a "Scale to Fit?" dialog box for large (or "Scale Up?" for small) models. A quick Yes or No upon loading the occasional large/small model would cover both the novice and expert users.

awhiemstra, 2015-07-14:
@Xisle I like that suggestion and will probably implement it that way for the next feature release. It provides a nice unintrusive option to do both behaviours and should be fairly simple to implement. The only thing you would lose is scaling precision when dealing with very large or very small objects. But maybe we can add an "apply transformation" or similar option in the context menu.

nallath, 2015-07-15:
The message object already has support for buttons, so should be relatively easy to implement.

BagelOrb, 2015-07-17:
I would rather see scale-to-fit as context menu option. 
Dialogs can disrupt workflow.

My other suggestion was related to the fact that a model might be saved in cm rather than mm. In sui a case some autoscaling is desirable.
Just posting it here cause it is an issue just as easily fixed at the same time...

SteveHeiles, 2015-07-19:
This should be selectable over the SCALEIcon. Maybe the main icon can turn red, if it is not fitting and then you can puch the fit icon to scale it.
It should never be done without any notification!
My last object was fitting in the diagonal but not in the width and I didn´t notice this autoscale. So it took me 11hours to finally notice that the object it too small. So currently I have again to use the old Cura software.

BagelOrb, 2015-07-19:
Indeed an option in the scale menu, which only becomes visible when the object is too large is a good idea. When an inexperienced user sees the object is too large, it will go into the scale menu and find the scale-to-fit option there automatically, without having to know about that option existing in the context menu,

Thanks, Steve!

luisonoff, 2015-07-26:
Just wanted to add my experience, until new version is released and I can test these changes.
Yesterday I opened this model (http://www.thingiverse.com/thing:745523/#files, version2) on Cura 15.06.2.  It should perfectly fit my printing bed without any scaling (it does so in Slic3r), but Cura decided to automatically and without notice scale it to 95%. Worst thing was that when I opened the scale menu, it said 100%, but that was not true, as I found hours later when checking dimensions in the scale menu, which show real ones (scaled 95%).

Maybe not related to this, but here it goes: I rotated this item 90º, on the scale menu: axis dimensions show on bed axis correct, but scale boxes do not apply to bed axis, but original axis of the model before rotating it. I believe this is not expected?  It made wierd results, as for example, when I changed Y axis scale, actually it was making a change in Z bed axis.

BagelOrb, 2015-07-27:
@luisonoff Suppose you set Z axis to 50%, then rotate it 45 degrees. If then the Z-scaling controls the scale of the new z, there would be no way to get the original z axis back to 100%, because you are scaling a diagonal axis. Every time you rotate all axes would have to say they have a scaling of 100%, which is not intuitive.

luisonoff, 2015-07-27:
True. I didn't think about that. Then I would suggest that this is shown or explained somehow, for example on the menu put BED AXIS and MODEL or BODY AXIS, or something like that.

BagelOrb, 2015-07-27:
See also https://github.com/Ultimaker/Cura/issues/192

closed 2015-08-24 by nallath

red2black, 2015-08-28:
I'm not convinced that just giving a warning of auto scaling is a complete solution. As has been discussed above it would be much better if this were optional and could be avoided altogether.
And when the scale tool is selected will this show 100% or the scaled?
I and my colleagues are having to roll back to a previous version of Cura or switch to other software to avoid this problem.
On a more philosophical note, is an issue resolved if the fix is not released?

nallath, 2015-08-28:
It will show the correct scale. The old cura will also auto-scale, but only if the object is a -lot- bigger. You can scale back to the original size. It will not scale back (eg; the auto scale is only done on model load)

I could add an option to the preferences that disables this behaviour (it would be on by default though).

BagelOrb, 2015-08-28:
"The old cura will also auto-scale, but only if the object is a -lot- bigger."
I think such behaviour is better.

It would be nice if scale-to-fit is a context menu option, i.e. right click model -> [Scale to fit]

nallath, 2015-08-28:
There ya go! It will be in the next release.