Ultimaker/Cura #146: "Infill Line Width" Setting appears to alter spacing, but not extrusion rate

CCS86, 2015-07-15:
Visually, it looks like this is happening in Cura 16.06.2.

With uniform print speeds, a .4mm nozzle and infill line width set to .5mm, I'm seeing correct extrusion on my perimeters, but under-extrusion on all infill (100% density)

BagelOrb, 2015-07-16:
Try loading the output gcode into the old cura. The new Cura visualizes based on polygon data sent directly from the CuraEngine, while the old Cura visualizes the actual gcode.

It may be the case that the nozzle width is used for displaying all polygons instead of the corresponding line width.

tl;dr : probably vizualisation issue

CCS86, 2015-07-16:
I never rely on visualization for width type analysis, because I'm used to programs like Repetier using a fixed width bead for visualization.

This is an observation from actual printing. I tuned flow to actually over-extrude in the perimeters, and stil saw a slight failure to bond adjacent beads in the 100% infill region.

BagelOrb, 2015-07-16:
What kind of printer do you have?
Please view the file in the old Cura, since it does perform width type visualization correctly.

CCS86, 2015-07-16:
Ultimaker Original.

[image]

CCS86, 2015-07-16:
I created a small test part for this to compare across slicers. It is a simple rectangle allowing for 3 perimeter laps at .4mm width, and a single .6mm infill bead, allowing for 15% infill overlap per side:

[image]

Even with "infill line width" set to .4mm, Cura 15.06.2 refuses to infill:

[image]

Forcing extra bottom layers gets a fill:

[image]

However, it is actually messy and sparse, despite looking good in the Cura visualization:

[image]

Slic3r seem to nail it:

[image]

I calculated the amount of extrusion from the infill section between Slic3r and Cura gcode and found a large difference:

0.18314mm extruded by Cura

0.33814mm extruded by Slic3r.

It appears to not just be a visualization issue, but exactly what I originally thought. Cura seems to be spacing the beads as if it was laying down .6mm beads, but not calculating extrusion volume for that .6mm bead.

As a separate issue, it refuses to fill these areas with "normal" infill, and even when forced with "bottom" layers, still enforces a 45* angle which makes for a very clumsy infill.

BagelOrb, 2015-07-16:
You are currently reporting on three issues:
1) incorrect line_width / line_spacing combo
2) no infill (visialized) in some unidentified cases (https://cloud.githubusercontent.com/assets/12439543/8725455/f1fb866c-2b99-11e5-898f-ba67500a7d47.png)
3) jagged infill lines which should have been combined into one long line

Next time, please file separate bug reports.

I have not yet found the origin of 1. The bug boggles me. Did you make the setting infill line distance visible?

I am currently working on 3, cause I know a way to improve the code.

I don't have enough info on 2. Can you load the corresponding stl into the old Cura and see whether it is a visualization issue?

CCS86, 2015-07-16:
Sorry, I discovered the other two as I diagnosed the first. I thought they could be related.

1) No, I am not specifying infill spacing (spacing and density are either/or)

2) Not a visualization issue. I loaded the gcode and no infill is laid on non- top/bottom layers.

BagelOrb, 2015-07-17:
Can you copy paste your configuration file? For Linux machines its at home/.config/cura/
It contains all settings, so that I can easily reproduce the bugs.

CCS86, 2015-07-17:
this guy?

[cura]
active_mode = 1
categories_expanded = ;infill;resolution
active_machine = Ultimaker Original Plus
recent_files = C:/Users/Admin/Desktop/3D/Models/Calibration Parts/Infill Calibration.stl;C:/Users/Admin/Desktop/3D/Models/My Designs/Oil Drain Cap.stl;C:/Users/Admin/Desktop/3D/Models/Artistic/Lost_Pleiade_Ready3_fixed.stl;C:/Users/Admin/Desktop/3D/Models/Artistic/Lost_Pleiade_Ready3.stl;C:/Users/Admin/Desktop/3D/Models/Artistic/Yoda Smooth.stl;C:/Users/Admin/Desktop/3D/Models/Artistic/Doggy Vase.stl;C:/Users/Admin/Desktop/3D/Models/Artistic/Julia_Vase_004_-_Bloom_-_Solid.stl;

[view]

[general]
window_width = 1904
window_height = 1122
version = 2

[backend]

BagelOrb, 2015-07-19:
Alright. It seems you haven't altered any settings, so those should be easily reproduced ;)

I'm having a feeling this issue is specific to UMO machines and other machines which don't work with volumetric measures in extruded filament. (Instead they work on how many cm of material is extruded; the machine doesn't know about the diameter of the filament.)

I will look into this.

CCS86, 2015-07-20:
Thank you.

So in our case, all the volumetric calculations have to happen in the slicer, right? (volume/mm filament, and volume of the extruded rounded rectangle)

We give Cura a nozzle width in the settings; assuming we haven't specified any modified widths in the new engine, does Cura choose a fixed width extrusion bead, or attempt to automatically control it like Slic3r?

BagelOrb, 2015-07-20:
I think we are misunderstanding each other. When you don't override a setting, the default or inherited value will be used. The engine will always use the default given by the settings file when no user defined setting has been supplied. The frontend will always provide the engine with settings, also based on the settings file, but also takes inheritance into account.

CCS86, 2015-07-20:
I understand that.

I was talking about the difference in gcode between a UMO and UM2. You made it sound as though the UM2 had moved to volumetric extrusion, which would be handled by the firmware, vs the UMO which (by default) does not work that way.

The second part of my question was (when not overridden by user settings) does the new Cura use a constant line width for all portions of the print (and if so, what size?), or does it vary line width to improve printing automatically?

BagelOrb, 2015-07-20:
As w.r.t. your first part: yes.

As to the second part: the frontend will use the same line width for all features when the suer didn't specify/override them. This is equal to the nozzle size.

CCS86, 2015-07-20:
Great, thanks for clarifying!

I'm curious about your thoughts on this logic for extrusion layout/calculation:

http://manual.slic3r.org/advanced/flow-math

BagelOrb, 2015-07-20:
It might be interesting to have a line spacing differing from the line width. The rest will come down to better inheritance functions in the front-end settings.
I should certainly implement this one time or another.

BagelOrb, 2015-07-20:
I cannot reproduce your original issue (1). I've tried with UMO+ and default settings, but skins seem normal. I've tried both in the 15.06 branch and in the master branch. Perhaps the issue has already been solved in 15.05.02 or in the next release (15.06.03).

2) has been verified.

3) is what I'm working on, but is kind of at a halt atm.

CCS86, 2015-07-20:
I am running 15.06.2, so #1 isn't fixed there as far as I can see.

You have manually increased infill line width and seen correct extrusion amounts to match the modified spacing?

BagelOrb, 2015-07-20:
Hmm I didn't see those configurations in your settings file, so I didn't try.

Now I've reproduced 1) as well.

There's work to be done!

printam3d, 2015-07-21:
Hello,

I have the same issue with the infill. Now I'm printing something on my UM2 with 100% infill and as shown in CURA 15.06.2 there are some parts of the object where the printer doesn't prints infill. This doesn't happen on every layer but on most of them.
[image]
[image]
[image]

BagelOrb, 2015-07-21:
Thanks for the info.
Clear picture!
If I may ask: why are you printing Lines support instead of ZigZag support?

BagelOrb, 2015-07-21:
(3) has been fixed :)

BagelOrb, 2015-07-21:
As for (1)
[image]
[image]

Perhaps the GUI hasn't updated the value of infill_line_distance.
What happens if you make the infill line distance setting visible and then unvisible again?

CCS86, 2015-07-21:
I don't think I've ever had an issue with it when the width setting is invisible, because it then is defaulting to a line width. The issue comes when specifying a wider line width; where the spacing is increased, but the same amount of filament is extruded.

BagelOrb, 2015-07-21:
Is it true that you make the infill line width larger and then make it invisible again?
That would explain it.
When you make a setting invisible it becomes inherited from its parent again ("Line Width") while perhaps there is a bug which causes the infill line spacing not to reflect this change.

Did you change settings which you made invisible again?

BagelOrb, 2015-07-21:
@printam3d Could you please share your model with me? Then I can check your case specifically.

CCS86, 2015-07-21:
It won't generate infill at all for me without forcing a bunch of bottom/top layers. Specifying 0.4mm infill line width, 0.6mm infill line width, and hiding that setting, all give me this:

[image]

printam3d, 2015-07-21:
So just to be clear, for my project, I didn't modify the width of the infill or any other part of the print. Everything was set on default which is 0.4
BagelOrb I can share one of the projects but only with you. Please tell me how cause I'm relatively new to this forum.
You also asked why am I printing lines instead of zigzag. Because this is the second print with Cura 15.06.02. Until now I used 15.04 and I didn't have had the time to test other settings so I used the ones that were available on 15.04 also. The projected from which I posted the picture was a total failure with 15.04 since the support structure created was very bad. The one on 15.06.02 was more complex and that is way I took the decision to go with 15.06.02 even without testing on other projects of a less importance.

Ghostkeeper, 2015-07-22:
These most recent problems are possibly related to Ultimaker/CuraEngine#223

BagelOrb, 2015-07-22:
I don't think so.

BagelOrb, 2015-07-22:
I've solved the no-infill-issue now as well (2)
No need for you model anymore ;)

Indeed in the old Cura this was solved by making the top/bottom distance huge, so that the whole print is covered with skin. The problem was that there was an overlap between the infill and the skin, which could remove small parts of infill. Now there is no more overlap between infill and skin in case the infill is (around) 100%.

I'm sorry; I asked why you printed lines instead of zigzag because I thought ZigZag was the default, which is not the case. Now ZigZag is default.

The original issue still stands. I think there is some issue with the settings handling in the front-end, which causes the infill_line_distance or the infill_line_width not to get updated properly.

BagelOrb, 2015-08-12:
(Whoops)
The original issue still stands. I think there is some issue with the settings handling in the front-end, which causes the infill_line_distance or the infill_line_width not to get updated properly.

BagelOrb, 2015-08-27:
My thoughts seem to have been correct.

This is currently fixed in a branch (per-object-settings) which will get merged to the master branch soon.

closed 2015-09-08 by BagelOrb

ryanbranch, 2017-07-30:
I believe this issue is happening again. Personally, when "Infill Line Width" is increased above the "Line Width" value, the infill spacing is altered but it still continues to extrude infill at the old line width. As demonstrated earlier in the thread, the rendered "Layers" view shows what the infill would look like if it were extruding at the proper width.

I "proved" this to myself by setting **line width to 0.4 mm** and **infill line width to 0.8 mm**. Then I made a solid rectangular print, setting the **infill to 100% with the "Lines" pattern**. The resulting infill was quite sparse, equivalent to the pattern one would expect with 0.4 mm infill line width and a 50% infill density.

I'm printing on a **Monoprice Maker Select v2** and running **Cura 2.6.1**. I'd be happy to provide any gcode, screenshots, or other information as requested.

CCS86, 2017-07-30:
I would say to go ahead and post all that stuff

ryanbranch, 2017-07-30:
[image]

[image]

Here are 2 screenshots of an object in Cura. The only things changed between the 2 are the Infill Percentage and Infill Line Width. Despite their difference in appearance, I believe these 2 items would both print as if they were the 50% infill, 0.4 mm wide lines

I also have associated gcode for both but it won't let me share it through this platform. I'd be happy to post it if anyone has suggestions for alternative places for that.

fieldOfView, 2017-07-30:
Note that setting infill density to 100% does not actually use infill, but instead uses "top" layers all the way to the bottom of the print. Could you try 99% instead?

Ghostkeeper, 2017-08-14:
When I try this myself (also removing the automatic 999999 bottom layers), I get the expected 100% infill. I get thicker lines in the infill which are appropriately spaced such that they align exactly to 100%. This was tested on Ultimaker 2 Extended+, for lack of the Maker Select definition of Ryan's.

In the second screenshot it looks from the layer view and the material estimates that this would print with the expected 100% infill as well. The lines are thicker in the layer view and they seem to align perfectly. Also, the second screenshot uses 79cm more material. Looks fine, right?

You should be able to upload the g-code here on Github if you first put it in a zip archive.