Ultimaker/Cura #147: Nozzle Size

JohannesK91, 2015-07-16:
Please give us the Nozzle Size back instead of editing the .json file. Having 5 Um2 Profiles is not that great. 

Thanks

nallath, 2015-07-16:
We're working on it. It's on the roadmap for next release.

BagelOrb, 2015-07-16:
Do you have 5 different printers with different nozzle sizes? I can't imagine!
I guess you want to edit the line width, not the nozzle size....

JohannesK91, 2015-07-16:
i mean this line    "machine_nozzle_size": { "default": 0.4 }in the .json,
and then i edit the wall width, thats the way i had done it 15.04 and now i do in 15.06.
But in 15.04 there was the nozzle Size in cura it self

But the easy way would be editing the machine_nozzle_size in cura and then set the wall line coun't and the width would be set automatic

BagelOrb, 2015-07-16:
machine_nozzle_size is really just the actual width of the whole in the nozzle.

If you want to adjust all line widths at the same time, just go to machine settings > quality > shell thickness > wall thickness > wall line width
Unhide only wall line width, and none of its children.
Change the wall line width, to change all line widths.

Indeed this settings should be named 'line width' and put somewhere else.

JohannesK91, 2015-07-16:
With the olson Block you change this hole to .8mm .6mm...

But if works only by editing the wall line width, then it should be named different, because its not only me    who didn't understand it

BagelOrb, 2015-07-16:
Yes indeed, I am working on it right now..

BagelOrb, 2015-07-16:
Note that the raft line widths are separately controlled - not by the general line_width setting.

BagelOrb, 2015-07-16:
Fixed on the master branch. This will be released in October.

closed 2015-07-16 by BagelOrb