Ultimaker/Cura #148: Cura 15.06.02 Crash in Windows 8.1

milkok, 2015-07-16:
Run Cura 15.06.02 first time,I have to add printer.Whether I choose any one of the options, click "Next", the software will stop working. If I click "Cancle",it won't stop.

awhiemstra, 2015-07-16:
Does it also crash when you choose the Ultimaker Original? If it does not, it might be the same bug as #96

Ghostkeeper, 2015-07-17:
I just tried this and couldn't reproduce it. I also use Windows 8.1.

I had an install of 15.06.01 already, so I went to AppData and deleted Cura's settings. Then I installed 15.06.02 and at startup it asked me if I wanted to add printers. I chose Ultimaker 2 and successfully added a printer.
So apparently, it doesn't always occur or doesn't occur on all Windows 8.1 machines.

milkok, 2015-07-18:
I found it ! There's something wrong in the json files. The "platform_texture" doesn't work.

awhiemstra, 2015-07-22:
That suggests it is indeed the same issue as #96, so closing this as duplicate.

closed 2015-07-22 by awhiemstra