Ultimaker/Cura #149: Cura not reading lock files when accessing serial port (linux)

axsdenied, 2015-07-16:
On linux Cura will access serial port without checking if it is use. This can have disastrous consequences.

To fix:
1. Cura should check if lock file exists before accessing the serial port, for example /var/lock/LCK..ttyACM0
2. If the file exists Cura should check if the PID of the process that created the file exists (in the first line of the file). 
If it does then Cura should not try to connect to the serial port as it is in use.
If it does not Cura should delete the stale lock file and proceed with port access.

omegahacker, 2016-07-14:
I can confirm the disastrous results, though I suppose I'm partially responsible because my own code doesn't create said LCK files.  I have MCU programming hardware that exposes a bunch of serial ports, and whenever Cura is running, they all started to go totally insane.  Figured out that Cura was spamming all of the ports constantly.  This is extremely BAD behavior even if it were to use the LCK files properly.

Also, there are various symlinks the kernel produces based on the device's USB properties, e.g. /dev/serial/by-id/usb-Malyan_System_LTD._Malyan_3D_Printer_Port_6D8915694855-if00, but I can't tell if the "standard" says that the locking application should name the lock after "chasing" the links.

Because I'm trading off between working on that hardware and working on my printer, I'm going to have to temporarily remove the USB serial printing plugin.....

BagelOrb, 2017-07-21:
Is this fixed now?

nallath, 2017-07-21:
Nope

Ghostkeeper, 2017-10-30:
Our project manager just removed this from our planning because it's more than 12 weeks old.

Ghostkeeper, 2018-05-04:
Yeah, on Windows it does that too.

ianpaschal, 2018-05-22:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-22 by ianpaschal