Ultimaker/Cura #150: Show settings as command line arguments in engine log

BagelOrb, 2015-07-16:
Often I get bug reports on the backend from people operating the frontend without knowing with which options they have called the backend.
Can you append the settings with which the backend has been called to the engine log?
Preferably in the correct form ("-s [setting]=[value]") so that I can easily reproduce a bug and/or eliminate the backend as the origin of the bug.

I'm also fine with being able to retrieve the exact settings with which the engine was called in another way.

nallath, 2015-07-16:
I've just added the command list to the error log as that might also show some extra info. Adding the entire list (in the correct form) will take a more work, as we now tell the engine where the .json with the settings is located.

BagelOrb, 2015-08-20:
This issue is about **settings**.

Perhaps this should be done in the engine...

BagelOrb, 2015-08-27:
[implemented]

closed 2015-08-27 by BagelOrb