Ultimaker/Cura #151: "CuraEngine.exe has stopped working" on loading STL (Cura 15.06.02)

Fuzionist1, 2015-07-17:
1. Opened a basic STL file in Cura 15.06.02 (Shared on my Dropbox - https://www.dropbox.com/s/k1giwgx0szb3caw/Fan_Baseplate.stl?dl=0)

2. Error pop-up "CuraEngine.exe has stopped working" (see attached Capture.png)
[image]

Notes:
- No error when opening in Cura 15.06.01
- STL was created in Fusion 360

awhiemstra, 2015-07-22:
This is a bug in the engine as far as I can tell. I get this backtrace:

#0  0x00000000004f8856 in std::vector<std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> >, std::allocator<std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > > >::size() const ()
#1  0x00000000004f7b94 in cura::Polygons::size() const ()
#2  0x0000000000501ecf in cura::CommandSocket::sendPolygons(cura::PolygonType, int, cura::Polygons&) ()
#3  0x0000000000523055 in cura::FffPolygonGenerator::sendPolygons(cura::PolygonType, int, cura::Polygons&) ()
#4  0x000000000051f568 in cura::FffPolygonGenerator::slices2polygons(cura::SliceDataStorage&, TimeKeeper&) ()
#5  0x000000000051e43a in cura::FffPolygonGenerator::generateAreas(cura::SliceDataStorage&, PrintObject*, TimeKeeper&) ()
#6  0x0000000000504313 in cura::fffProcessor::processModel(PrintObject*) ()
#7  0x000000000050157a in cura::CommandSocket::connect(std::string const&, int) ()
#8  0x00000000005368af in main ()

awhiemstra, 2015-07-22:
Actually, that was with the master version from the engine. That version has known issues unrelated to this. Testing with the released version of 15.06.02 on Windows 8.1 I get no crashes. Any chance you can list settings that are not set to their default values?

Fuzionist1, 2015-07-23:
The error was consistent and reproducible before, but not it is not.  You are probably right - it was probably a setting - or combination of settings, and I don't know what they were anymore.  Thanks for your help.

closed 2015-07-23 by Fuzionist1