Ultimaker/Cura #153: Autoresize when importing large stl object

SteveHeiles, 2015-07-18:
When importing a large object Cura 15,6,2 is auto-resizing the object so that it fits on the print plate...

BagelOrb, 2015-07-18:
True

closed 2015-07-18 by BagelOrb

SteveHeiles, 2015-07-18:
?Sorry but this is a bug!!!!!
The Object was to large to be printed in the width but not in the diagonal.
At least the system should tell you that it is doing an autoresize!!! Or there should be somewhere a possibilty to switch it of. In the olde version 04 this is still ok.
Some people are caring about the size of an object being printed out.

BagelOrb, 2015-07-18:
Fair enough. The way you first posted it didn't sound like a bug report. That's why I responded the way I did.

This issue is currently discussed athttps://github.com/Ultimaker/Cura/issues/142

When reporting an issue, please confirm to the scheme employed in the above issue.