Ultimaker/Cura #154: Cura crashes at the start

dani16494, 2015-07-19:
Everytime i open Cura on my Mac, the following error appears:
------------------------------------------------------------------------------------------------------------------
Version: 15.06.02
Platform: Darwin-14.3.0-x86_64-i386-64bit
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "/Applications/Cura.app/Contents/Resources/cura_app.py", line 10, in <module>
  File "cura/CuraApplication.pyc", line 176, in run
  File "UM/Application.pyc", line 246, in setActiveMachine
  File "UM/Signal.pyc", line 96, in emit
  File "cura/CuraApplication.pyc", line 503, in _onActiveMachineChanged
  File "cura/BuildVolume.pyc", line 142, in rebuild
TypeError: can't multiply sequence by non-int of type 'float'
------------------------------------------------------------------------------------------------------------------

My Mac:
Yosemite 10.10.3
i5 2,4GHz
8GB DDR
Intel HD Graphics 3000 512 MB

nallath, 2015-07-21:
Did you change anything in your setting .json files? I'm not able to reproduce the bug and it should not even be possible for this to occur in the first place.

dani16494, 2015-07-22:
Not shure but i don't think so.
I also downloaded it a few times again and reinstalled it but it still does not work.
Cura 15.04 works fine but the new cura version is a lot better i think...
It stopped working after i worked with DMX_JOKER but i think this should not be the reason why Cura does not work. 
I would be really glad if i could work with Cura 15.06.2 again.
Do you need any log-files or something?

nallath, 2015-07-22:
Yeah, that would be appreciated. I can simply put a check in the code that ignores this error, but I'd rather find out what causes it (and fix that)

awhiemstra, 2015-07-22:
From the error, it looks like there is something that uses a comma (,) for decimal separator instead of a dot (.). Since we have not had any other reports about this, it suggests you have a custom profile that is invalid.

murphy2712, 2015-07-22:
I fixed this issue by deleting (or renaming) the old Cura preferences folder: ~/.cura

dani16494, 2015-07-22:
Where can i find this preferences folder? I think it could really be a setting file because in the background the window appears and there are all my preferences.

closed 2015-08-14 by nallath