Ultimaker/Cura #155: M25 missing at the end of gcode?

luisonoff, 2015-07-19:
I do not know if this is intentionally or a bug, but 15.04 Cura used to add a M25 (SD Print Pause) at the end of my gcode but 15.06 Cura does not. 
Is this an activable feature or am I supposed to add it at the EndGcode of machine settings?

nallath, 2015-07-21:
This is not a feature at this point, aside from adding it to the end-gcode in the machine settings.

closed 2015-07-21 by nallath

luisonoff, 2015-07-21:
Thank you for the answer :)

BagelOrb, 2015-07-23:
M25 used to mean SD Print Stop for an UM2. It was added because of the profile string which was added at the end of the gcode. However, this is not done any more, so M25 is not necessary any more.

luisonoff, 2015-07-23:
What is that "profile string which was added at the end of the gcode"?

With my printer I still need to add M25 at the end, or print never stops and keeps counting time, etc. 
I just added the M25 to the endgcode at json file of machine settings in new cura and it works perfect.

BagelOrb, 2015-07-23:
Your solution is the correct one for your specific issue.

The profile string was used for the old Cura to recover settings from a gcode file...