Ultimaker/Cura #156: Old UI?

sarf2k4, 2015-07-20:
I somewhat missed the old UI, it provides much cleaner looks and responsiveness to me. Any way to have the old ui back but with the new things from the new versions such as plugin supports and other parameters?

daid, 2015-07-20:
Nope, we changed too much for that. We don't want to maintain two versions.
One of the reasons to make the new code is to clean up the mess I made (code wise) in the past few years. We're working on the responsiveness issue as well.

Old versions are still, and will remain, available as always.

sarf2k4, 2015-07-20:
I guess I have to wait for the new UI to somewhat stabilized. I gotta say still lack of "user-friendliness" =)

closed 2015-07-21 by nallath