Ultimaker/Cura #158: crash Version: 15.06.02

SEADEMON, 2015-07-20:
Version: 15.06.02
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 75, in __init__
  File "c:\dev\uranium\UM\Application.py", line 355, in loadMachines
  File "c:\dev\uranium\UM\Settings\MachineSettings.py", line 90, in loadValuesFromFile
  File "c:\dev\uranium\UM\Resources.py", line 50, in getPath
FileNotFoundError: Could not find resource ('tam_series1.json',) in 2

nickthetait, 2015-07-20:
What does it take to reproduce this crash? Just open Cura? Did this version of Cura work previously but something changed to make it crash?

SEADEMON, 2015-07-21:
15.06.01 some time crashes. But work. 15.06.02 try start but can't. All
time crash

nallath, 2015-07-21:
Cura saves the latest active machine upon startup. In your case this is the tam_series1. However, the json file that it requires for this has been removed, which must be done by the user.

closed 2015-07-21 by nallath