Ultimaker/Cura #159: fan minimum turn on voltage (or %) and custom ramp curve would be exceedingly helpful

aviphysics, 2015-07-21:
Feature Request:

Some way to set a custom fan ramp curve (either vs z height or time) with a minimum "%" turn on voltage, would be very helpful. This would be especially useful for people using crossflow fans, as the platform has a strong focusing effect on the airflow during the first few millimeters of the print. It would also help with fans that require a substantial voltage before they will spin-up.

Thank you for your work on Cura.

Sincerely,
Aviphysics

Ghostkeeper, 2015-07-22:
I think that something like a spin-up voltage should be implemented in the firmware of the machine, in an ideal world. The machine firmware knows that its fans require some extra voltage to overcome the static friction, and how long this voltage needs to be maintained.

Of course, Cura may take this task off the printer manufacturer's hands by faking it via g-code. I guess that a fan spin-up time could be added to the machine settings, which outputs "M106 S255" to the g-code any time the fan speed changes from 0 to x>0, and then after the specified amount of seconds sets the fan speed to the proper value.
Is that the sort of thing you are looking for?

aviphysics, 2015-07-22:
For fan spool up and spool down rate, I think it would indeed be useful to be able to set a maximum "acceleration", in order to prevent rapid cooling of the print head.

However, I was also kind of thinking of it being useful to be able to set a non-linear ramp curve in some way. In the current implantation, from Z == 0 to "Fan Full on at Height" the fan speed setting is increased linearly with each layer. It is sometimes the cast that a linear adjustment is not ideal. This is especially the case when using a crossflow fan, where the print bed has a substantial focusing effect on the airflow during the first few mm of the print. It would be useful to be able to have a more complex ramp curve (just having a few linear segments, instead of just one linear ramp, would probably do the trick.)

daid, 2015-07-22:
But airflow is also no linear with the PWM value.

(And a fan-kickstart is already implemented in Marlin)

BagelOrb, 2015-07-22:
I don't see why you would want a different curve for different materials. Perhaps you need more cooling for a different material, but that would just be a different place on the curve.

Ideally the gcode contains only instructions for the amount of cooling, while the firmware uses that value to calculate the fan speed needed.

That being said, it might very well be the case that the amount of cooling needed over the first few layers shouldn't decrease linearly. However, I have no reason to think it deviates much from linearity.

You arguments pertain solely to the fan speed, not the amount of cooling needed. Therefore I think this is a firmware issue.

closed 2015-07-22 by BagelOrb

daid, 2015-07-22:
Also, the ramp-up was mostly implemented because you don't want cooling on the first layer (so it sticks better), but you want cooling on all others. However, when you suddenly turn the fans to full-blast, most printers PID temperature controllers reacted very badly to the change in heating power needed.