Ultimaker/Cura #162: Save model does not honor scale changes 15.06.02

CountMurphy, 2015-07-21:
I have tested this with both the Windows and Linux clients. I have a model which I accidentally scaled incorrectly when I created it. The model in question is 76.4 mm on the x axis. My typical workflow is to load the model into cura to position, rotate and scale it 76.7 mm, save the stl and then send it to replicatorg to export to my makerbot. With the new release of cura, any model I save as an stl retains the models original dimensions. For example I can import my model, scale to 76.7 mm (on the x axis), save the model to stl and open that newly saved model with cura. The loaded model will still have the dimensions of 76.4.

this was tested with cura 15.06.02

nallath, 2015-07-21:
Cura also re-centers an object when it loads it in the current implementation. Would it be a problem if that is also saved?

CountMurphy, 2015-07-21:
I wouldn't think that would be an issue.

awhiemstra, 2015-07-22:
This is being overhauled for the next feature release. Unfortunately, that work cannot be backported to 15.06. I will try and see if I can implement a quick fix for 15.06, but otherwise I am afraid you will need to wait until the next feature release.

nallath, 2015-07-22:
Saving the transformed mesh would work as a quick fix right?

awhiemstra, 2015-07-22:
Yes, that is what needs to be done for 15.06.

CountMurphy, 2015-07-22:
Awesome. Thanks for looking into this. Much appreciated.

closed 2015-09-15 by nallath