Ultimaker/Cura #163: Multiple Part Positioning

marcus2tts, 2015-07-21:
Trying to place multiple objects on the build platform in 15.06 is just a nightmare. I'm using Win8.1. First of all, please don't automatically place every new object in the middle of the plate and then forcing what's already there out of the way. Let me put it where I want it.

When duplicating an object, duplicate it in the same orientation as it's currently positioned in. Right now it duplicates and places in the original orientation in the center of the platform; pushing anything that already occupies that space out of the way.

I've purposely had to move things off the platform to still have some control over them. But when i go to move them back onto the platform, as i click on it, it flies out of the window; refocuses the window, and then there is no way to move the object back onto the platform as the pick center of the object is nowhere near the center of the part anymore. 

There has to be a way to remove all the history of the part's transformations while keeping them where they are. Barring that, there should be a numerical transformation display of each object, with the ability to punch in the numbers you want.

luisonoff, 2015-07-22:
I just wanted to say that I experienced this same thing today of having multiple parts, clicking one to move it and then window moves, changes pick center but you only realize when you move it a bit because it goes crazy further than waht you expected...

FuzzySadist, 2015-07-28:
Ditto here, I was getting ready to write this up.  
[image]

nallath, 2015-07-28:
This is a known issue. We're working on it. It was caused by another fix.

mdklapwijk, 2015-07-28:
And on the note of the duplicate option, please bring back the "how many duplicates" dialog.

nallath, 2015-07-28:
Please make a seperate issue for every request. We cant keep track otherwise.

nallath, 2015-08-20:
Part of this is fixed for the next release; You can disable the automatic push free.

BagelOrb, 2016-04-26:
Fixed

closed 2016-04-26 by BagelOrb