Ultimaker/Cura #169: Setting-changes register at inconsistent events (or don't register at all)

fieldOfView, 2015-07-22:
The numeric values in the Print Setup column register at inconsistent keyboard/focus events.

Steps to reproduce with 15.06.02 on Windows 7:
* Start Cura and load a model.
* Open the Material panel.
* Place the cursor at the end of the temperature field, and press backspace removing the last digit of the value. 
-> Cura starts slicing. This is unexpected, because I am not yet done entering the value
* Press '0', adding a digit to the value
-> Cura does not start slicing again. This is expected, because I may still not be done editing the value.
* Press Tab, putting the focus on the next value. Or press enter to confirm the value.
-> Cura still does not start slicing. This is unexpected, because the value has changed. Also there is no way to force Cura to start slicing with the new value. Not even clicking back into the Temperature field, and pressing enter.

* Close Cura, and start it back up again.
-> Note that Cura has stored the value with the last digit removed, and ignored the digit that was added back in.

All of this is inconsistent with common practices in most GUIs and toolkits I know, and it will lead to reports of settings not working (eg issue https://github.com/Ultimaker/Cura/issues/168).

I strongly suggest not starting the slicer while entering values, but only once the user is done editing the value. I think the best way to do this is to record the value of a field when the field gets the focus, and to compare it to the value when the field looses focus (be it by Tab, by Mouse, or otherwise) or when the user presses Enter to confirm the value. Storing the new value, and starting a new slice job should only happen if those two values differ.

awhiemstra, 2015-07-22:
We originally only started slicing when confirming the setting change. We got several issue reports that this was unexpected so changed it to react to text changes. We could change it back but then you end up with new reports that people change settings and nothing happens. So I would rather just address the actual issue, setting changes not triggering a reslice. Oddly enough it seems limited to just 0 from some quick testing, other numbers properly trigger a reslice. For certain things this is logical, as 0.10 is equal to 0.1 as far as a computer is concerned. For other things this is weird, as changing the number of brim lines from 2 to 20 should really trigger a reslice.

fieldOfView, 2015-07-22:
> We got several issue reports that this was unexpected so changed it to react to text changes

Wut? Users are weird...

fieldOfView, 2015-07-22:
Reverting https://github.com/Ultimaker/Uranium/commit/9b678862dfaced54fd0bad04f633e42b423dd2ef does fix the issue for me.

awhiemstra, 2015-07-22:
> Wut? Users are weird...

Yes, they are. ;)

fieldOfView, 2015-07-22:
At least part of the problem seems to be that ```value``` never changes here:
https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/qml/UM/Settings/SettingTextField.qml#L19
```text``` seems to be compared to the ````value``` that was loaded when the application started.

nallath, 2015-08-14:
Sorry it took so long. I've changed it. Nice catch.

closed 2015-08-14 by nallath

fieldOfView, 2015-08-14:
It's going to be interesting to see if that fixes all of this issue.

nallath, 2015-08-14:
In so far i've been able to see it did, but I might have missed some.