Ultimaker/Cura #170: Crash handler not handling python exceptions on Windows

fieldOfView, 2015-07-22:
I still get the "Errors in Cura.exe" error, instead of a new Crash handler. Cura 15.06.02, Windows 7.

Uninstalled 15.06.02, uninstalled 15.06.01 (just to be sure), made sure there was no 15.06.02 folder left in Program Files, redownloaded todays 15.06.02 and installed it. After playing around in layer view, I still get the "Errors in Cura.exe" dialog instead of the Crash handler.

https://ultimaker.com/en/official-ultimaker-news/view/16751-cura-150602?page=1#reply-112952

fieldOfView, 2015-08-01:
Same with 15.06.03

awhiemstra, 2015-08-03:
Yeah, I found out the exceptions don't bubble down all the way, instead they hit the Qt event loop and are printed to console. I have a fix for it but it requires a bit of testing to see what the real impact is.

awhiemstra, 2015-08-10:
The fix has been pushed to master and will be used in the next feature release.

closed 2015-08-10 by awhiemstra