Ultimaker/Cura #173: Cura Crash on Startup: invalid result type

escentrix, 2015-07-23:
I'm running Arch Linux and have been trying to get Cura working again since the updates. I started with version 15.06.01 and it installed fine, and even started up fine the first time. After I restarted the program, however, I got the follow error:
...
2015-07-22 20:54:24,786 - INFO - Loaded plugin CuraEngineBackend
TypeError: invalid result type from AvailableMachinesModel.roleNames()
Aborted (core dumped)

I recently got 15.06.02 installed and it is doing the same thing. I've tried manually removing user and system files and reinstalling, but nothing seems to be working.

awhiemstra, 2015-07-23:
Hmm, I recall seeing something about this on the PyQt mailing list, but I am not sure what causes it. What version of PyQt/Qt are you using?

escentrix, 2015-07-23:
Looks like both 4.11.4-1 and 5.5-1 are installed:

extra/pyqt4-common 4.11.4-1
extra/pyqt5-common 5.5-1
extra/python-pyqt4 4.11.4-1
extra/python-pyqt5 5.5-1
extra/python2-pyqt4 4.11.4-1
extra/python2-pyqt5 5.5-1

I looked and they both have different dependencies in my system, but is having them both a problem?

awhiemstra, 2015-07-23:
No, it is a change in PyQt5.5 that introduced the error. Just uprgaded my Arch system to 5.5 and getting the same, should have a fix soonish.

escentrix, 2015-07-23:
Excellent, thank you!

awhiemstra, 2015-07-23:
This fixed things when using PyQt 5.5:

https://github.com/Ultimaker/Uranium/commit/041ae0ee89b9af5a53c20589a51c70a7a9e761d7

Unfortunately I ran into another issue with custom geometry for UI elements, which seems to be a regression in PyQt 5.5 . For now I implemented a workaround, but it means all elements that had their corners cut off are now rendered as rectangles.

escentrix, 2015-07-24:
That fixed the error for me. I can live with non-rounded corners I think! Thank you again for your help.

closed 2015-07-24 by escentrix

balwierz, 2015-07-31:
I have just uploaded cura-bin to AUR which includes the patch.

escentrix, 2015-07-31:
Awesome! Thank you very much for that balwierz! Just tried it out, everything is working great! You're package install seemed to have fixed another issue I was having with CuraEngine crashing in the background.