Ultimaker/Cura #175: Cura Crash on startup on UbuntuKylin15.04

supergis, 2015-07-23:
On UbunbtuKylin15.04,which as same as Ubuntu15.04.
Run: python3 cura_app.py
will get error:
QWidget: Must construct a QApplication before a QWidget

I am using Cura-build to construct this project.

nallath, 2015-07-23:
Have you tried downloading the precompiled cura? This error can only occur when an interface element is shown before the QApplication is loaded, which means that you're either not using the latest version of the code or made changes of your own that cause this.

awhiemstra, 2015-07-23:
Is there no log output? Because in that case it most likely is failing to load important bits of the code. For starters, you want to set PYTHONPATH to $builddir/inst/lib/python3/dist-packages .

MiguelAngelLV, 2015-07-24:
I have the same error.

```bash
e with 'origin/master'.
link@Hyrule:~/dev/Cura$ python3 cura_app.py
QWidget: Must construct a QApplication before a QWidget
Abortado
```

I use the lastest code from github, the master branch

supergis, 2015-07-25:
I use the lastest code from github, the master branch. and set PYTHONPATH. but the problem exist still.
not found any log output.

supergis, 2015-07-25:
I try:
try:
    import cura.CuraApplication

    #app = cura.CuraApplication.CuraApplication.getInstance()
    #app.run()
except Exception as e:
    print(e)
    #import cura.CrashHandler
    #cura.CrashHandler.show()'''
'''
get result:
'''
No module named 'UM'
'''

supergis, 2015-07-26:
In my computer,the Uranium for Python located: /usr/local/lib/python3/dist-packages/UM
set PYTHONPATH to /usr/local/lib/python3/dist-packages
run as above.
"cura.CuraApplication.CuraApplication.getInstance()" will get error:
```
libGL error: core dri or dri2 extension not found
libGL error: failed to load driver: vboxvideo
OpenGL Warning: XGetVisualInfo returned 0 visuals for 0000000002faf810
OpenGL Warning: Retry with 0x8002 returned 0 visuals
OpenGL Warning: glXGetFBConfigAttrib for 0000000002faf810, failed to get XVisualInfo
OpenGL Warning: XGetVisualInfo returned 0 visuals for 0000000002faf810
OpenGL Warning: Retry with 0x8002 returned 0 visuals
段错误 (核心已转储)
```
I am using UbuntKylin15.04 in VirtualBox based MacOSX Yosemite 10.10.

shushengxsss, 2015-07-26:
I have the same error, and my  system is newly installed ubuntu 15.04.

nallath, 2015-07-27:
@supergis That seems like a sperate issue. Could you post it in it's own issue?

lotuspaperboy, 2015-08-03:
I'm seeing the same problem to on Ubuntu 14.04. I've tried setting PYTHONPATH etc. but I am still seeing a crash. Has anyone made any progress with this?

awhiemstra, 2015-08-04:
I have just updated the crash handler to handle import errors and related more gracefully.

@lotuspaperboy If you're running with standard Ubuntu PyQt packages you will get an exception about an unsupported PyQt, since we have a hard dependency on PyQt 5.4, due to known crash bugs in older versions.

nallath, 2015-08-14:
Have you made any progress with this?

closed 2015-08-26 by nallath