Ultimaker/Cura #180: Cura 15.06.02 Engine crashes on Windows 8.1

ThatJCB, 2015-07-25:
Just drag and drop the STL file and the engine crashes.
The application is completely unusable for me and will revert to 15.04.

nallath, 2015-07-27:
Could you please post the log file and a more detailed flow of what you did / where trying to do (What file did you try to slice? What are the specs of your machine?)

ThatJCB, 2015-07-27:
Workflow was simple. 
- Save STL file from Blender. 
- Open Cura 15.06.02. 
- Drag the STL file from Explorer on to the bed image in Cura.
The model is tiny, so small I cannot see it even zoomed it. Just a grey square shadow!
- Do nothing else
- Wait about 1 second
- The engine crash error appears.
I'll post the logs from my home machine tonight.

I have tried several models. All work perfectly with version 15.04

I am on my work machine today, so I thought I'd try it on this.
This does not crash as often but never finishes calculating:
- Save STL file from Blender. 
- Open Cura 15.06.02. 
- Drag the STL file from Explorer on to the bed image in Cura.
The model is tiny, so small I cannot see it even zoomed it. Just a grey square shadow!
- Do nothing else
- Wait and wait and wait and still waiting. 
It never finishes calculating and the Save to disk button remains grey.

This is the log file from TODAY. Not crashed but also not working.

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Warning! Overlapping faces: face 0 and face 5.
1; 0;0.00021205
Warning! Overlapping faces: face 0 and face 5.
1; 0;0.00021205
Warning! Overlapping faces: face 0 and face 8.
1; 0;0.00021205
Warning! Overlapping faces: face 2 and face 5.
1; 0;0.00053694
Warning! Overlapping faces: face 5 and face 8.
1; 0;0.00021205
Warning! Overlapping faces: face 5 and face 6.
1; 0;0.00021205
Warning! Overlapping faces: face 8 and face 5.
1; 0;0.000722298
Warning! Overlapping faces: face 9 and face 5.
1; 0;0.00021205
Warning! Overlapping faces: face 9 and face 5.
1; 0;0.00021205
Warning! Overlapping faces: face 9 and face 8.
1; 0;0.00021205
Unregistered setting retraction_count_max
Unregistered setting retraction_extrusion_window
Unregistered setting material_switch_extruder_prime_speed
Unregistered setting material_switch_extruder_retraction_speed
Unregistered setting machine_switch_extruder_retraction_amount
Slicing model...
Unregistered setting meshfix_keep_open_polygons
Unregistered setting meshfix_extensive_stitching
Layer count: 0
Sliced model in 0.000s
Generating layer parts...
Unregistered setting meshfix_union_all_remove_holes
Unregistered setting meshfix_union_all
Generated layer parts in 0.000s
Finished prepareModel.
Unregistered setting multiple_mesh_overlap
Unregistered setting magic_polygon_mode
Stopping process because there are no layers.
Unregistered setting retraction_extra_prime_amount
Wrote layers in  0.00s.
maxObjectHeight : 50
Print time: 0
Filament: 0
machine_gcode_flavor = UltiGCode
machine_gcode_flavor = 1
Total time elapsed  0.00s.

ThatJCB, 2015-07-27:
Machines:
Home that crashed
Lenovo laptop
Intel i5
8GB RAM
256GB SSD
Dock using Lenovo Thinkpad USB 3.0 for use with 24" wide screen monitor
Windows 8.1 set to auto-update

Work that never completes:
Dell E7440 Latitude laptop with touch screen
Intel i5
8GB RAM
256GB SSD
Dock using Lenovo Thinkpad USB 3.0 for use with dual 24" wide screen monitors
Windows 8.1 set to auto-update

ThatJCB, 2015-07-27:
I undocked the laptop and ran on just the laptop screen to rule out the USB 3.0 dock as the problem.
I got exactly the same symptoms.
Tiny model and never finishes calculating.

I also tried changing the scale, but it is stuck or reverts to 100%.

ThatJCB, 2015-07-27:
I tried someone else's STL file and that worked.
It may only be STL files exported from Blender that do not work with 15.06.02
I'll do some more testing.

ThatJCB, 2015-07-27:
Found the problem on my work machine.
15.06.02 does not honour the scale setting from Blender.

I'll start a new issue because I do not know if they are related.

I create my models in mm and set the Metric scale in Blender to 0.001 to change meters to millimeters.
If I change that back to a scale of 1.0 before exporting the STL file then it works in Cura 15.06.02.

I will not know until this evening if that fixes my engine crash on my home machine.

ThatJCB, 2015-07-27:
See also issue #191

BagelOrb, 2015-07-27:
tl;dr : there are some bugs when the model is way too small.

I don't think this is worth fixing, since the error only occurs for unreasonably small models.

nallath, 2015-07-27:
Could you also post the log file? That should contain the log info from the GUI.

ThatJCB, 2015-07-27:
Sorry, the log file is empty!

Here are the entire contents after the engine crash!

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.

ThatJCB, 2015-07-27:
I have saved the STL file again to the correct size. This definately fixes it.

Therefore the only problem with 15.06.02 is that it does not handle tiny files. Stating the obvious but it should trap this error and not crash.

A simple message saying the object is too small would be sufficient.

nallath, 2015-07-28:
There should also be a log file. This is the engine log, which is different. But yeah, we should add such a message.

ThatJCB, 2015-07-28:
What is the log file called and where is it?
I did a search of the Cura program files and sub-folders for anything with the word log in the name and found nothing.

ThatJCB, 2015-07-28:
Can I suggest a message similar to the following:

"The imported model is too small for Cura to process!
Please re-size your model in your drawing package using one unit = 1mm."

What would be even better is if Cura offered a dialogue box to manually enter a scale to resize the imported model.
No guessing, no auto-sizing,  just give people the chance to make a manual correction if the model is too small.  Perhaps even if it is too large.

Thanks

nallath, 2015-07-28:
Check the readme. We've updated it with the log info.

We won't add a dialogue box for every model. This would impact the 99% of the users that don't have this issue far to much.

BagelOrb, 2015-07-28:
Closing this issue, since the crash is avoided when the feature request in https://github.com/Ultimaker/Cura/issues/191 is implemented.

Note that the bug would still be there.

closed 2015-07-28 by BagelOrb