Ultimaker/Cura #181: Default end retract 6 mm but start feed only 3 mm (ultimaker_original.json)

aviphysics, 2015-07-25:
The default total retraction at the end of a print is 6 mm, but the feed at the start of the print is only 3 mm. ( this is in the ultimaker_original.json ). Is this intentional?

nallath, 2015-07-31:
I don't see that in the start / end g-code.

daid, 2015-07-31:
But I do.

nallath, 2015-07-31:
I only see an retraction of 1 mm (G1 E-1 F300). Definitely not 6mm.

daid, 2015-07-31:
M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 **E-5** X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
M84 ;steppers off\nG90 ;absolute positioning

nallath, 2015-07-31:
Thats not in the end code at the moment; 
"M104 S0\nM140 S0\n;Retract the filament\nG92 E1\nG1 E-1 F300\nG28 X0 Y0\nM84"

nallath, 2015-08-14:
This also seems to be the case with the old cura, as it's a copy of that start / end code. Anyone know why that is done?

BagelOrb, 2015-08-17:
Fore reference: 
in ultimaker_original.json the start and end codes are as follows.

start:
------
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 *E3* ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000
;Put printing message on LCD screen
M117 Printing...

end:
------
M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 *E-1* F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 *E-5* X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
M84 ;steppers off
G90 ;absolute positioning

I think the correct way to solve this is to increase the initial primed amount / unretraction. 
[ implemented ]

closed 2015-08-17 by BagelOrb