Ultimaker/Cura #182: Combine everything missing

cody42, 2015-07-25:
In cura 15.06 and later the "combine everything" option(s) is/are missing.

BagelOrb, 2015-07-27:
meshfix_union_all
"Union Overlapping Volumes"
Settings > Preferences > Fixes

cody42, 2015-07-27:
It is missing, see:
[image]
There is no fixes page in the settings and it is not listed in the fixes section of the macine section.

Version: Cura 15.06.02, prebuilt from ultimaker.com
OS: Windows 7 x64

BagelOrb, 2015-07-27:
I'm sorry. I should have said it is fixed in the master branch, so the next major release has them.

Due to the fact that we cannot make new translations of the settings, we cannot introduce new settings in the current release.

Note that there is no fixes tab because you have no settings from that tab visible.

closed 2015-07-27 by BagelOrb

tzminversion, 2015-09-03:
Im using Cura 15.06.03 an I still can choose "Combine everything".

I am constantly going back to 15.02.1 because features have been ripped out in the new version unfortunately :( shame really.

BagelOrb, 2015-09-04:
Why don't you go back to 15.04 ?

The feature is in the current beta release.

tzminversion, 2015-09-05:
Probably because I didn't know there was a 15.04 ;) Tnx!