Ultimaker/Cura #183: 15.06.2: Not Slicing

CCS86, 2015-07-26:
Loaded a model, and could see Cura attempting to slice while ticking through all the settings I wanted. But once I finished, it didn't even try. 

Engine log:

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Unregistered setting retraction_count_max
Unregistered setting retraction_extrusion_window
Unregistered setting material_switch_extruder_prime_speed
Unregistered setting material_switch_extruder_retraction_speed
Unregistered setting machine_switch_extruder_retraction_amount
Slicing model...
Unregistered setting meshfix_keep_open_polygons
Unregistered setting meshfix_extensive_stitching
Layer count: 527
Sliced model in 0.717s
Generating layer parts...
Unregistered setting meshfix_union_all_remove_holes
Unregistered setting meshfix_union_all
Generated layer parts in 0.234s
Finished prepareModel.
Unregistered setting multiple_mesh_overlap
Unregistered setting magic_polygon_mode
Unregistered setting alternate_extra_perimeter
Unregistered setting ooze_shield_enabled
Generated inset in 2.699s
Generating support areas...
Generated support areas in -0.000s
Unregistered setting fill_perimeter_gaps
Generated up/down skin in 3.120s
Unregistered setting wipe_tower_distance
Unregistered setting retraction_extra_prime_amount
Unregistered setting extruder_nr

Cura.log:

Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 10
Exception in thread Thread-84:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\libArcus\python\Arcus\__init__.py", line 228, in run
  File "c:\dev\libArcus\python\Arcus\__init__.py", line 267, in _receiveNextMessage
TypeError: can't concat bytes to bool

CCS86, 2015-07-26:
I quit and reloaded Cura, hen loaded the same model. The settings were where I left them, and it sliced successfully. Then I made one change: infill=lines -> zigzag. It began slicing then hung about 75% through:

[image]

Cura.log:

Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 155, in event
  File "c:\dev\uranium\UM\Event.py", line 149, in call
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Backend\Backend.py", line 128, in _onSocketError
AttributeError: 'UnknownProtocolError' object has no attribute 'errno'

CCS86, 2015-07-26:
Reloaded again, and hung once more, with a different error:

Exception in thread Thread-18:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\libArcus\python\Arcus\__init__.py", line 228, in run
  File "c:\dev\libArcus\python\Arcus\__init__.py", line 267, in _receiveNextMessage
TypeError: can't concat bytes to bool

nallath, 2015-07-27:
It could be that the zigzags is broken. I've made some fixes to the engine & frontend how some of these are handled. This will be in the next version.

CCS86, 2015-07-27:
Hi Nallath, I have found the "line 155" error happening, even with lines infill pattern. It seems to happen after doing something to trigger repeated slicing starts (moving the part around, changing settings).

BagelOrb, 2015-07-27:
"
 'UnknownProtocolError' object has no attribute 'errno'
"

sounds like a meta-error 
D:

nallath, 2015-07-28:
That error is already fixed. It has nothing to do with this.

nallath, 2015-08-14:
Does this error still occur (with 15.06.3)?

CCS86, 2015-08-14:
It seems to be working correctly now. I'll keep an eye on it. Thanks!

closed 2015-08-14 by nallath