Ultimaker/Cura #184: Lacking options: infill after perimeters and initial layer line width

luisonoff, 2015-07-26:
At new Cura 15.06.2 I miss these options:
- Infill after perimeters
- Initial layer line width

I guess they are still to be implemented?

nallath, 2015-07-27:
@BagelOrb Are they still in the engine? If so, it should be easy to add them again.

BagelOrb, 2015-07-27:
They are not in the engine any more/yet.

Since now there is no more line_width option, but instead infill_line_width, wall_line_width etc. how would you propose to do the initial layer line width? Have tons of settings for each feature? Or have one setting which all features (infill, walls etc.) use?

I'm thinking of adding more options for deciding the order of the features (infill, walls etc.), which should at least contain the possibility for printing the infill after the perimeters.

luisonoff, 2015-07-27:
I would like to have a box called something like "first layer line width", if you put a width there, for example 0.6mm, it does full first layer (all features) with that width; and if you put a %, like 150%, it would increase each feature by that percentage for first layer.
Maybe this is not very userfriendly or should be explained on a small note when you hover it.

BagelOrb, 2015-08-25:
Infill after perimeters has been implemented once again.

I'm still not sure about the other option. Your suggestion is just unmaintainable. I probably wouldn't ever find out that the input box would work that way if I didn't know already.

I think we should choose between a percentage or a static value.

luisonoff, 2015-08-25:
Then I would choose the % option. So user can increase (decrease) every feature line width by the same factor.

BagelOrb, 2015-08-25:
I was leaning more toward the one-size-fits-all, since the line width doesn't scale linearly with adhesive properties.

Now I'm thinking it is best to have separate line width options for the different features. Then the outer wall could inherit from the normal wall thickness while the inner walls and skin inherit from line_width_layer_0.
The CuraEngine shouldn't be concerned with the fact that there are too many settings - that's a frontend issue.

tzminversion, 2015-09-05:
My opinion: Initial layer Width in (%) should be used as before.

It would also be nice if "initial layer height" could be 0.1 and the rest of the layers 0.2. I tried this and from what I understood from feedback is that "initial layer height" is a multiple of "layer height".

BagelOrb, 2015-09-07:
I'm planning on including an initial layer thickness per feature, even though it will introduce quite a lot of settings. That way you can specify a larger initial layer thickness for the bottom skin infill - giving better adhesion - while having normal initial layer thickness for the outer wall, resulting in a more accurate first layer.

@tzminversion Initial layer height can already be configured. You misunderstood something; there is no such relation between initial layer height and normal layer height.

tzminversion, 2015-09-07:
@BagelOrb thanks for the feedback. I will experiment then with the initial layer height then. I remember an UM developer telling me that the initial layer height can not be less than the normal layer height. This may have applied to an older version perhaps, or I am completely mistaken... I will try ;)

khudko83, 2015-12-27:
BagelOrb, you wrote that Infill after perimeters has been implemented once again in Cura 15.06, but I still can't find this feature in the 15.06.02 and 15.06.03.
It is critical for me to use 15.06.02 and 15.06.03.
Was this feature added or not to mentioned versions of software?

fieldOfView, 2015-12-30:
I don't know about this specific setting, but do check out the later betas. A lot has happened since 15.06.03:
http://software.ultimaker.com/Cura_closed_beta/15.10/
Or better yet: try the 2.1 branch!

BagelOrb, 2016-01-04:
Why do you insist on using 15.06? It is very outdated and full of bugs!
15.06 and 15.10 are alpha/beta versions of Cura 2.
Cura 2.1 will be released quite soon; the first official release of Cura 2.

khudko83, 2016-01-04:
I just have my Malyan M180, which is capable with 15.06 only due to nessasity to convert gcode to x3g! And x3g plugin only works with 15.06
When do you plan to release Cura 2?
Moreover, in 15.06.04 I have the problem of underextruded scar, while the whole object is printed with no underextrusion. And only scar laced have lack of plastic... I tried to set a retract to 0, but it didn't help... Can it be issue of ver. 15.06?

BagelOrb, 2016-01-04:
You can find other beta releases for Cura 2 here:
http://software.ultimaker.com/Cura_closed_beta/15.10/

The official release will be soon. You will have to wait at least a couple of weeks.

Where did you get the x3g plugin?

khudko83, 2016-01-04:
Here is Cura 15.06 remaked by Malyan with integarated M180 support and x3g plugin - http://wiki.malyansys.com/doku.php?id=cura_15
I tried to use this plugin with Cura 15.10, but always received an error... I also can't choose another (new) printer in 15.10, just those, which are predefined in the Cura from very beginning... I tried to add a new printer by coping .json file to the appropriate Cura folder, but it changes nothing! This method doesn't work in Cura 15.10
Maybe you can somehow assist me in order to use Cura with Malyan M180? Thanks in advance!

BagelOrb, 2016-01-04:
Malyan should have submitted their edits to the Cura code base instead of leaving the fork in their own repository.

We can't guarantee the compatibility if they don't contribute their edits to our code base.

khudko83, 2016-01-04:
I fully agree with you, but I have a problem now and I see that no one can help me... That is why I'm trying to solve it by my own with old version of Cura...
But still I will be very grateful to you if you can look at those Malyan version and advise me how to implement their x3g plugin in Cura 15.10 together with adding a new printer...
Thanks!

BagelOrb, 2016-01-04:
Can you link to the source code of the Malyan version? I cannot find it.

khudko83, 2016-01-04:
As I can understand there is no source code for Cura. I ran exe file (http://www.malyansys.com/downloads/Cura150603-malyan-win.exe) and it is actually an achived folder with Cura (no installation required). And there are plugin folder and .json file for Malyan M180 in the folder... I tried just to use those files with new Cura versions, but with no success...

khudko83, 2016-01-04:
Here is Malyan repository with GPX (x3g plugin) source code - https://github.com/MalyanSystem/GPX

BagelOrb, 2016-01-04:
If you make a pull request with the json file we can see if we can include it for future releases.
There has to be a plugin which enables the GPX. If you make a pull request with that one as well we can see whether we can include it.

khudko83, 2016-01-04:
How can I make pull requests for plugin and printer? As I can nderstand there is no easy way at this time to adapt the plugin from 15.06 to 15.10 to let me use the printer with newer Cura? And did you change the procedure of adding a new printer in 15.10? I have no options to choose some Another printer from Add new printer dialog...

BagelOrb, 2016-01-04:
15.10 is an outdated version.

In the current master I can just Add Printer > Other > [select printer]

You would have to place the json file in the correct place first though..

khudko83, 2016-01-04:
Where can I get this Master version? And can you help to use x3g from 15.06, made by Malyan?

fieldOfView, 2016-01-04:
> And can you help to use x3g from 15.06, made by Malyan?
Can't you ask Malyan instead? You paid them for the printer, and it is their addition to the software. We know nothing about the x3g plugin, and since the do not share their changes to Cura we would have to manually reverseengineer and check everything to see what they have done.

khudko83, 2016-01-04:
Many thanks for your priceless advise! Will not disturb you anymore in this case...

Ghostkeeper, 2016-01-05:
I made a pull request of your Malyan file, @khudko83. A pull request is meant to be a request from an outside user to include code in the program, so the code should've been ready to be included, as it were. A developer is then to look at the code and decide whether or not to include it in the project.

I've now made the translation for you from the 15.06 file to the 2.1 file, and requested it to be included in the next version (pending review).

khudko83, 2016-01-05:
Ghostkeeper, many-many thanks for your kian assistance!

khudko83, 2016-01-05:
Ghostkeeper, can I also ask you to help me to create a pull request for "gcode to x3g converting in new Cura" to allow saving files in Makerbot format? Malyan also implemented it in 15.06 (as a plugin), but doesn't want to do it again now in newer versions...
The repository of GPX (used to make a x3g pluging in 15.06) is here - https://github.com/MalyanSystem/GPX
Details of how it works in Cura 15.06 are here:
http://wiki.malyansys.com/doku.php?id=cura_is_not_able_to_save_as_x3g_format
http://wiki.malyansys.com/doku.php?id=gpx_for_translate_gcode_into_x3g_file
Here is Cura 15.06 with integrated by Malyan x3g plugin - http://wiki.malyansys.com/doku.php?id=cura_15
I have also attached achive with plugin forled from Cura 15.06 Malyan version (please rename .TXT file to .ZIP to open it)
[Cura_15.06.03.txt](https://github.com/Ultimaker/Cura/files/78396/Cura_15.06.03.txt)
I afraid to do it in wrong way, if I will try to open request by myself...
Many thanks again!

BagelOrb, 2016-01-06:
You shouldn't be scared to do the pull request in the wrong way. We will review the request and if you have done anything wrong we will point that out so that you have the chance to change it before we accept the pull request.

khudko83, 2016-01-06:
BagelOrb, ok, thanks! I will try to do it right!

fieldOfView, 2016-01-06:
@khudko83 FYI, here's a video about the basics of how Pull Requests work:
https://www.youtube.com/watch?v=YTbRzhQju4c
It's about wordpress, but it should teach you how to make simple changes and make a pull request from that. If you want to do more than simple changes, learn how to use git on your computer to clone the repository and push changes back to github.

Ghostkeeper, 2016-01-06:
@khudko83 If you contact me at r.dulek (at) ultimaker (.) com I'll be able to help you with that plug-in. This issue thread has gone off-topic for quite a while so it's probably better to keep it in a separate channel. I've just done some related work on Cura with backend output writers, and I'd gladly help with fixing that plug-in. I'm not sure if it's appropriate to include it in Cura by default though (I'm all for it), but that's up to debate with our Cura team.

rcmonitor, 2016-09-06:
So, what about Initial layer line width?
Would it be implemented in the first place?
I really miss this setting in Cura 2.1.3, since there is no other way to affect adhesion.

Ghostkeeper, 2016-09-07:
A workaround is to set the initial layer height a bit higher and adjust your print bed a bit higher to compensate. It has the same effect.

The issue hasn't been planned yet for any release, but it is a valid issue.

rcmonitor, 2016-09-07:
> A workaround is to set the initial layer height a bit higher and adjust your print bed a bit higher to compensate
> It has the same effect.
No, absolutely not.
Initial layer line width changes both offset between lines and amount of filament to be extruded, so the lines are completely valid, just wider, thus dramatically increasing adhesion.
On the other hand, when you make incorrect homing (i.e. set bed to 0.10 when software is set to 0.20 on the first layer) you are making lines wider for some unknown amount, but offset between lines stays the same, so one line take some square of the other and squeezes itself aside, which is totally wrong.

BagelOrb, 2016-09-07:
What type of printer do you have?

Here at Ultimaker people don't seem to have that much trouble with bed
adhesion.

<https://github.com/Ultimaker/Cura/issues/184#issuecomment-245344424>, or mute
the thread

rcmonitor, 2016-09-07:
I've gor reprap prusa i3.

> people don't seem to have that much trouble with bed
adhesion
All people?
With all printing materials?
On all print sizes?
With all adhesion materials?
...Really doubt it.
[http://googleitfor.me/?q=ultimaker+first+layer+(delamination%7C%7Cwrapping)](url)

> Here at Ultimaker

So, what is your point?
This software may be used with ultimaker printers only?
Then just declare it in documentation to this software, so owners of other models not to bother you with their issues.

BagelOrb, 2016-09-07:
Whow, hold your horses.

I'm just trying to figure out what the priority of this issue is wrt other
features. And yes, that is the priority a feature has to *us*. Supporting
other printers has lower priority to us.

I still haven't got enough information to make the call on this feature.

We do accept pull requests, so if you really want this feature you can
contribute to Cura.

The fact that we make our software open to other printers is a courtesy.
I expect this courtesy is returned in the form of contributions to Cura so
that together we can make 3D printing better for everyone.

Ghostkeeper, 2016-09-08:
> line width changes both offset between lines and amount of filament to be extruded

Ah, true. Well, I agree that it's neater to change the actual line width. And thicker lines will also cause the nozzle to move more slowly while still theoretically covering the same surface area per time.

BagelOrb, 2017-07-21:
Implemented: initial layer line width percentage

closed 2017-07-21 by BagelOrb